Entware/Entware #255: Package libopenssl / hard/soft floats incorrect

SuperHirmu, 2019-03-25:
Presently in current Packages List for armv7soft platform are two entries for libopenssl:

- libopenssl - 1.1.1a-2 (uses hardfloats, breaks everything)
- libopenssl - 1.0.2n-1 (uses softfloats, is the correct version)

opkg update ; opkg upgrade leads to soft-floats libopenssl being upgraded with non-functional version, which breaks - among other things - node_legacy, wget.

Package list used by my router: http://bin.entware.net/armv7sf-k2.6/Packages.gz

zyxmon, 2019-03-25:
This is not a problem with hard/soft float. It is a problem with opkg system.
You need to reinstall wget - it may be tricky, since wget is used to download packages. You need a working wget copy in yor path.
As for node_legace - it is missing in repo. I'll compile and upload it. I've tested - it is working. The new version does not depend on libopenssl and uses internal one.
Many packages, that use libopenssl need to be manually reinstalled.

zyxmon, 2019-03-25:
node_legacy added to repo and working
```
root@ws880:/tmp/home/root# opkg install node_legacy
Installing node_legacy (v0.10.48-2) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/node_legacy_v0.10.48-2_armv7-2.6.ipk
Installing libuv (1.24.1-1) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/libuv_1.24.1-1_armv7-2.6.ipk
Configuring libuv.
Configuring node_legacy.
root@ws880:/tmp/home/root# npm -version
2.15.1

```

closed 2019-03-25 by zyxmon