Entware/Entware #260: BZIP2 not working after last upgrade

vfiuchcikicshuusrch, 2019-04-01:
BZIP2 not working after last upgrade,
I have backup /opt
old version worked ok, new version show only "bzip2 not found"

zyxmon, 2019-04-02:
1. What is your hardware?
2. Is busybox installed?
3. Try to fix the issue with `/opt/lib/opkg/info/busybox.postinst` in this case
4. If you have installed bzip2 packages - try to reinstall it with `opkg remove --autoremove bzip2; opkg install bzip2`

Please write here your hardware and entware installation (standard or alternative) and console output of commands.

vfiuchcikicshuusrch, 2019-04-02:
1. Asus RT-AC56U (wifi router + fresh tomato firmware)
2. busybox into firmware already installed 1.25.1

after install busybox from entware into /opt/bin it not startup, but path is /opt/bin:/opt/sbin + $PATH.
I don't know how to fix that.

zyxmon, 2019-04-02:
Please answer all questions and show logs. Is busybox entware package installed. If not - 4 is your solution.

vfiuchcikicshuusrch, 2019-04-02:
log in attach.
which log you need ?
I think problem with linked library....

what I need fix in busybox.postinst ?

[log.txt](https://github.com/Entware/Entware/files/3033323/log.txt)

zyxmon, 2019-04-02:
I am going to fix this bug today and will write here new instructions. Thanks for the log.

vfiuchcikicshuusrch, 2019-04-02:
Thanks you too! I will be waiting!

zyxmon, 2019-04-02:
`opkg install --force-reinstall bzip2` will fix problem.

closed 2019-04-02 by zyxmon

vfiuchcikicshuusrch, 2019-04-02:
`opkg update` first.