Entware/Entware #261: Update package For SMCRoute

smcgrath12, 2019-04-02:
Package request

--- SMCRoute is a tool to enable Multicast routing in Linux by statically adding Multicast groups and routes to Linux kernel.

The current version in Entware is 2.0.0, which is close to 5 years old. The latest version available from the developer is 2.4.4.

- Package is in active development (at least, not abandoned for 5 years or more),
- I'm ready to test added package on my {platform} and get a feedback about it.

ryzhovau, 2019-04-02:
This package came from [upstream](https://github.com/openwrt-routing/packages/blob/d82d299a27c0c3e475c8e5be08ccb12dc3cbc433/smcroute/Makefile) with no changes.

Please, [ask ](https://github.com/openwrt-routing/packages/issues) maintainers to update it. The new version will arrive here during next sync.

closed 2019-04-02 by ryzhovau

smcgrath12, 2019-04-02:
Hi Ryzhovau, I am sorry, but I am not sure what I need to do.. Contact SMCRoute developers? What is the meaning of “package came from upstream”? I am ignorant, this is my first package update request to Entware.

ryzhovau, 2019-04-02:
Sorry, it's not an ignorance, let me explain that.
Entware uses OpenWrt feeds like [packages](https://github.com/openwrt/packages), [routing](https://github.com/openwrt-routing/packages) or [telephony](https://github.com/openwrt/telephony) and some unique feeds like [rtndev](https://github.com/Entware/rtndev) or [golang](https://github.com/Entware/entware-go) to fill our repo.

We keep our unique package update and sync with OpenWrt feeds once per months or so. We *won't* add any significant changes to OpenWrt packages or our sync process will run out of control. 

Ask OpenWrt guys to update package you want, just follow a link I provided before.

smcgrath12, 2019-04-02:
Thank you. I submitted the update request as per your instructions. Hopefully they get time to update it. Fingers crossed!