Entware/Entware #263: Python dependency broken for Asus RT-AC68U?

jackyaz, 2019-04-03:
```Installing python (2.7.15-5) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/python_2.7.15-5_armv7-2.6.ipk
Installing python-base (2.7.15-5) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/python-base_2.7.15-5_armv7-2.6.ipk
Installing libffi (3.2.1-3) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/libffi_3.2.1-3_armv7-2.6.ipk
Installing libbz2 (1.0.6-5) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/libbz2_1.0.6-5_armv7-2.6.ipk
Configuring python-base.
Configuring libffi.
Collected errors:
 * opkg_download: Failed to download http://bin.entware.net/armv7sf-k2.6/libbz2_1.0.6-5_armv7-2.6.ipk, wget returned 8.
 * opkg_install_pkg: Failed to download libbz2. Perhaps you need to run 'opkg update'?
 * opkg_install_cmd: Cannot install package python.
```

Package looks like it should be http://bin.entware.net/armv7sf-k2.6/libbz2_1.0.6-5a_armv7-2.6.ipk instead

zyxmon, 2019-04-04:
> Package looks like it should be
Package was changed (PKG_VERSION=5a). You need to run package update to update package index, that has the correct version.

jackyaz, 2019-04-04:
Ok, I'll advise users to do so (and have added opkg update to my installer script)

closed 2019-04-04 by jackyaz