Entware/Entware #264: Gerbera does not start from init.d

robolaf, 2019-04-04:
Hi,
Gerbera is the successor of Mediatomb. However, in Gerbera the -P option is not supported anymore(while it is in Mediatomb). Therefore the script
[https://github.com/Entware/rtndev/blob/master/gerbera/files/S90gerbera](url)
does not work anymore as the program raises an error because of the wrong parameter.
Could someone find another solution to run Gerbera at start with Entware?
Thanks and regards.

ryzhovau, 2019-04-26:
Thanks for report.

closed 2019-04-26 by ryzhovau

crisc82, 2019-08-20:
Have this been fixed? Because it was not working for me
```
/opt/etc/init.d/S90gerbera start
/opt/etc/init.d/S90gerbera check
 Checking gerbera...              dead.
```

trying to start -> error -> check the log again:
```
2019-08-20 18:21:16    INFO: Loading configuration from: /opt/etc/gerbera//config.xml
2019-08-20 18:21:16    INFO: Checking configuration...
2019-08-20 18:21:16   ERROR: No such file or directory: /root/.config/gerbera (errno: 20)
```
edit the config.xml, find for "/root/.config/gerbera" replace with "/opt/etc/gerbera"
tried to start the S90gerbera -> error -> check the log again:

```
2019-08-20 18:25:10    INFO: Loading configuration from: /opt/etc/gerbera//config.xml
2019-08-20 18:25:10    INFO: Checking configuration...
2019-08-20 18:25:10    INFO: Setting filesystem import charset to ANSI_X3.4-1968
2019-08-20 18:25:10    INFO: Setting metadata import charset to ANSI_X3.4-1968
2019-08-20 18:25:10    INFO: Setting playlist charset to ANSI_X3.4-1968
2019-08-20 18:25:10    INFO: Configuration check succeeded.
2019-08-20 18:25:10    INFO: About to migrate metadata from mt_cds_object to mt_metadata
2019-08-20 18:25:10    INFO: No data will be removed from mt_cds_object
2019-08-20 18:25:10    INFO: Migrated metadata - object count: 0
Exception raised in [src/server.cc:123] upnp_init(): upnp_init: UpnpInit failed
2019-08-20 18:25:10   ERROR: LibUPnP could not bind to socket.
2019-08-20 18:25:10    INFO: Please check if another instance of Gerbera or
2019-08-20 18:25:10    INFO: another application is running on port TCP 0 or UDP 1900.
2019-08-20 18:25:10   ERROR: upnp_cleanup: UpnpUnRegisterRootDevice failed
Exception raised in [src/server.cc:258] shutdown(): upnp_cleanup: UpnpUnRegisterRootDevice failed
```
so i also installed the mediatomb and checked the config.xml of both programs side by side. i modified gerbera config until i was able to start the service. this is my config:

```
<?xml version="1.0" encoding="UTF-8"?>
<config version="2" xmlns="http://mediatomb.cc/config/2" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="http://mediatomb.cc/config/2 http://mediatomb.cc/config/2.xsd">
  <server>
    <port>51500</port>
    <interface>br0</interface>
    <ui enabled="yes" show-tooltips="yes">
      <accounts enabled="no" session-timeout="30">
        <account user="gerbera" password="gerbera"/>
      </accounts>
    </ui>
    <name>Gerbera</name>
    <udn>uuid:b34a1deb-a0f5-4840-a467-cfbb3fb132da</udn>
    <home>/opt/etc/gerbera</home>
    <webroot>/opt/share/gerbera/web</webroot>
    <alive>1800</alive>
    <storage>
      <sqlite3 enabled="yes">
        <database-file>gerbera.db</database-file>
		<synchronous>off</synchronous><!-- default=off | normal | full -->
        <on-error>restore</on-error>
        <backup enabled="no" interval="6000"/><!-- default=no | yes -->
      </sqlite3>
    </storage>
    <protocolInfo extend="yes" samsung-hack="yes"/><!-- For PS3 support change to "yes" -->
    <custom-http-headers>
      <!-- Samsung needs it -->
      <add header="transferMode.dlna.org: Streaming"/>
      <add header="contentFeatures.dlna.org: DLNA.ORG_OP=01;DLNA.ORG_CI=0;DLNA.ORG_FLAGS=01500000000000000000000000000000"/>
    </custom-http-headers>
    <extended-runtime-options>
      <mark-played-items enabled="no" suppress-cds-updates="yes">
        <string mode="prepend">*</string>
        <mark>
          <content>video</content>
[... 74 lines cut ...]
    <profiles>
      <profile name="oggflac2raw" enabled="no" type="external">
        <mimetype>audio/L16</mimetype>
        <accept-url>no</accept-url>
        <first-resource>yes</first-resource>
        <accept-ogg-theora>no</accept-ogg-theora>
        <agent command="ogg123" arguments="-d raw -o byteorder:big -f %out %in"/>
        <buffer size="1048576" chunk-size="131072" fill-size="262144"/>
      </profile>
      <profile name="vlcmpeg" enabled="no" type="external">
        <mimetype>video/mpeg</mimetype>
        <accept-url>yes</accept-url>
        <first-resource>yes</first-resource>
        <accept-ogg-theora>yes</accept-ogg-theora>
        <agent command="vlc" arguments="-I dummy %in --sout #transcode{venc=ffmpeg,vcodec=mp2v,vb=4096,fps=25,aenc=ffmpeg,acodec=mpga,ab=192,samplerate=44100,channels=2}:standard{access=file,mux=ps,dst=%out} vlc:quit"/>
        <buffer size="14400000" chunk-size="512000" fill-size="120000"/>
      </profile>
    </profiles>
  </transcoding>
</config>
```
the entered in my web-browser on http://192.168.1.1:51500/ , set my autoscan dir.
Start my Samsung Tv.  The (dlna) movies are running fine!

So my conclusion is: **gerbera should come with a properly made configuration file like mediatomb have**. the auto-generated config file is big and hard to configure by a regular user if he don't the mediatomb config example.