Entware/Entware #265: Entware packages instant dead

digimon04, 2019-04-06:
For some reason, I had to push the power button of my RT-AC68U to reboot it. I have done this several times in the past without any issues. However, with the recent one, after the router starts, all Entware packages (i.e. Pyload, Transmission, Aria2) installed are dead. I try to start them manually but only result in instant dead.

> admin@RT-AC68U-6258:/tmp/home/root# /opt/etc/init.d/S88transmission start
> Starting transmission-daemon... done.
> admin@RT-AC68U-6258:/tmp/home/root# /opt/etc/init.d/S88transmission check
> Checking transmission-daemon... dead.
> admin@RT-AC68U-6258:/tmp/home/root#

> admin@RT-AC68U-6258:/tmp/home/root# /opt/etc/init.d/S51pyload start
> starting pyLoad
> Traceback (most recent call last):
> File "/opt/bin/pyload", line 27, in <module>
> from getopt import getopt, GetoptError
> ValueError: bad marshal data (string size out of range)
> admin@RT-AC68U-6258:/tmp/home/root#

> admin@RT-AC68U-6258:/tmp/home/root# /opt/etc/init.d/S81aria2 start
> Starting aria2c... done.
> admin@RT-AC68U-6258:/tmp/home/root# /opt/etc/init.d/S81aria2 check
> Checking aria2c... dead.
> admin@RT-AC68U-6258:/tmp/home/root#

Swap and Diversion run as usual. Could any one help me to troubleshoot it or tell me how to restore them? Thanks

zyxmon, 2019-04-06:
If you are using stock Asus firmware - it does not support scripts on reboot. Filesystem can become corrupted. Pid files from previous services runs can prevent new runs.

digimon04, 2019-04-07:
The router is currently running Asuswrt-Merlin. The scripts survive reboot in the past but not this time. When I start the entware package manually, they are dead instantly.

zyxmon, 2019-04-07:
There were a lot of changes in the last update, openssl was updated from 1.0.2 to 1.1.1. Some packages need to be reinstalled manually. Have you tried - https://github.com/Entware/Entware/wiki#note-about-packages-upgrade
You also may need to run binaries withoit scripts from init.d to find what is wrong.

digimon04, 2019-04-15:
Problem solved by upgrading packages. Many thanks.

closed 2019-04-15 by digimon04