Entware/Entware #266: trouble wget'ing upgrades/installing fresh on R7000 dd-wrt

8VhyJEYUH71CGe60, 2019-04-06:
Hi,
I tried rolling the latest upgrade last week and something broke.
Now I can't even install fresh entware (sh generic.sh).
My wget is dd-wrt's busybox.
My issue is that wget seems to stall mid download from bin.entware.net and pkg.entware.net (after waiting get generic error (code 1)).
editing generic.sh to curl instead of wget works, but then i get stuck on "opkg install entware-opt" (due to underlying wget).
Is there currently an issue with the server(s)? tcpdump seems to suggest the servers stop responding out of the blue.

zyxmon, 2019-04-06:
If your busybox version 1.26.2 - it has broken wget, that downloads corrupted files from nginx powered servers. You need to use another firmware in this case.

8VhyJEYUH71CGe60, 2019-04-06:
I apologize I did not have the foresight to include the version number (because I already troubleshot the issue you are talking about :))
BusyBox v1.30.1 (2019-04-03 20:01:01 CEST) multi-call binary.

zyxmon, 2019-04-06:
There are no server problems now.

zyxmon, 2019-04-06:
You can manually download and unpack busybox (Entware version), create a symlink /opt/bin/busybox -> /opt/bin/wget and set your PATH=/opt/bin:$PATH and try again

dnmTX, 2019-04-06:
@zyxmon  any way to substitute `wget` with `curl` when downloading packages or it's all baked in the `opkg` app? I couldn't find any additional scripts so i'm assuming it is.

8VhyJEYUH71CGe60, 2019-04-06:
Hi,

> 
> 
> You can manually download and unpack busybox (Entware version), create a symlink /opt/bin/busybox -> /opt/bin/wget and set your PATH=/opt/bin:$PATH and try again

Did it. 
stalled on "opkg install entware-opt" sub-process
"wget -q -O /opt/tmp/opkg-f74VXI/libgcc_7.4.0-8_armv7-3.2.ipk http://bin.entware.net/armv7sf-k3.2/libgcc_7.4.0-8_armv7-3.2.ipk"

(To show I'm not a (total) brainlet)
which wget
/opt/bin/wget

wget --help
BusyBox v1.30.1 () multi-call binary.

zyxmon, 2019-04-06:
You need working wget in your PATH. All the requests to http://bin.entware.net/armv7sf-k3.2/libgcc_7.4.0-8_armv7-3.2.ipk" returned 200 today (server logs).

8VhyJEYUH71CGe60, 2019-04-06:
> 
> 
> You need working wget in your PATH. All the requests to http://bin.entware.net/armv7sf-k3.2/libgcc_7.4.0-8_armv7-3.2.ipk" returned 200 today (server logs).

Am I missing something? (serious question)

echo $PATH
/opt/bin:/bin:/usr/bin:/sbin:/usr/sbin:/jffs/sbin:/jffs/bin:/jffs/usr/sbin:/jffs/usr/bin:/mmc/sbin:/mmc/bin:/mmc/usr/sbin:/mmc/usr/bin:/opt/sbin:/opt/bin:/opt/usr/sbin:/opt/usr/bin

which wget
/opt/bin/wget

ls -la /opt/bin/wget
lrwxrwxrwx    1 root     root            16 Apr  6 21:05 /opt/bin/wget -> /opt/bin/busybox

dnmTX, 2019-04-06:
@8VhyJEYUH71CGe60  just FYI. I did test it on my end to just download that package with `wget` to some test dir of mine and all went well.I think the problem is somewhere on your end and not in general so why don't you try with `nvram erase && reboot` and go from there.
[image]

8VhyJEYUH71CGe60, 2019-04-06:
> 
> 
> @8VhyJEYUH71CGe60 just FYI. I did test it on my end to just download that package with `wget` to some test dir of mine and all went well.I think the problem is somewhere on your end and not in general so why don't you try with `nvram erase && reboot` and go from there.
> [image]

I'm sure the problem is on my end, but that doesn't prove it works. The wgets sometimes work and sometimes don't  on my end - sometimes it stalls on packages before libgcc_7.4.0-8_armv7-3.2.ipk, sometimes after

zyxmon, 2019-04-06:
I see no problems with installing or downloading packages from my NAS that has the same armv7 Entware version.

zyxmon, 2019-04-06:
Some strange behaviour was observed from users that have connections through some "transparent proxies". Changing ISP (to 4G for example) helped in this cases.

8VhyJEYUH71CGe60, 2019-04-06:
> 
> 
> Some strange behaviour was observed from users that have connections through some "transparent proxies". Changing ISP (to 4G for example) helped in this cases.

Correct. Successfully installed w/ USB tethering.

8VhyJEYUH71CGe60, 2019-04-08:
It *did* work. Today I got the same issues w/ 4G tether. wget froze on "opcodes" dependency for gcc.
I will continue troubleshooting my router. 
Is it possible that am being flagged by the servers?

zyxmon, 2019-04-08:
Nobody is blocked on servers now.

8VhyJEYUH71CGe60, 2019-04-09:
Can't reproduce the issue for troubleshooting w/ wireshark.

closed 2019-04-09 by 8VhyJEYUH71CGe60

8VhyJEYUH71CGe60, 2019-04-10:
> 
> 
> You can manually download and unpack busybox (Entware version), create a symlink /opt/bin/busybox -> /opt/bin/wget and set your PATH=/opt/bin:$PATH and try again

FYI, no issues at all when using entware version wget (wget-ssl).
When I change back to busybox wget (of dd-wrt and of entware), or use wget-nossl the problems return.

If anyone is having similar issues in the future they can try to bootsrap entware by manually unpacking wget-ssl and by using it instead. (alternative to manually unpacking: install through opkg -> pkill wget when stuck -> install through opkg. this cowardly way might get you through all the dependencies a little by little.)