Entware/Entware #268: Package request: gocryptfs

wutr, 2019-04-10:
Package request
gocryptfs
---
Two years ago in entware-ng issue [#576](https://github.com/Entware/Entware-ng/issues/576) a request was made to include gocryptfs and/or CryFS. Not long afterwards CryFS was included, but gocryptfs was not.

Granted, both packages aim to achieve a similar goal. However, after experimenting with both for a reasonable time I feel gocryptfs may be the better solution for some users. During my (completely unscientific) tests gocryptfs was 3x faster in both read and write scenarios, because if compiled properly it can use the kernel to access hardware encryption accelerators ([/dev/crypto](https://github.com/cryptodev-linux/cryptodev-linux)) via OpenSSL. gocryptfs has much less overhead, for example ([provided by the author of gocryptfs](https://nuetzlich.net/gocryptfs/comparison/#disk-space-efficiency)), CryFS uses 1485MB to store the Linux 3.0 kernel source, whereas gocryptfs uses 494MB.

- [x] this one got some new killer feature(s),
- [x] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I'm ready to test added package on my {Synology DS216J} and get a feedback about it.

ryzhovau, 2019-04-22:
I see no reason to support two similar packages in repo, so we'll picked "classic" `cryfs`, sorry.
Also, libopenssl doesn't use `/dev/crypto` to keep our repo firmware-independent.

closed 2019-04-22 by ryzhovau

pgrm, 2022-02-17:
@ryzhovau a reason could be, if somebody stored their backups encrypted with gocryptfs on an object storage for instance and wanted to move them back to a new NAS. Oftentimes a NAS doesn't exist in isolation and data might exist in other places encrypted with other technologies, and this would be the reason to support alternative tools additionally to CryFS. 

Just to offer some additional perspective. Sure there are always other alternatives to move or convert the data.