Entware/Entware #270: CUPS printing is broken

davygravy, 2019-04-15:
First, thank you AR and Zyxmon for continuing to do this work.  It is appreciated.

It appears that CUPS printing functionality is broken.   This "issue" may have many components, but I felt this was a good place to start.

After installing CUPS, modifying its config file (cupds.conf and cups-files.conf), setting up proper users (nobody), etc.,  attaching a printer and working for over 1 week now trying to troubleshoot it, I've been unable to get anything to print via CUPS.

Known working:
-  HP USB printer tested, works via p910nd, so it is known as functional
-  installing CUPS:  Install goes fine, along with the dependent packages.
-  CUPS server runs, log level is set to debug, webinterface (:631) functions partly

Know not functional/problems/missing:
-  all basic lpr printing from the command line : literally nothing prints [ error : lp: Unsupported format 'text/plain' ]
-  missing components (printer test page, for example, and many other items) missing from proper
-  libcupsppdc.so.*, libcupsmime.so.*, libcupscgi.so.*  [I read your changelog and see that these were intended to merge with libcups.  libcupsimage.so.* IS present, but not the three listed above]
-  it appears to me that a large portion of cups-filters is missing : the older packages that I have seem to be much larger
-   error log full of "Filter failed" messages

Unknown: 
- were there some missed dependencies?
- is Ghostscript fully functional?
- is the Ghostscript-fonts-std package missing and causing problems?
- is there a problem with the function of HPLIP-full?  [ I suspect YES, as the older 3.16 version interacts very differently with CUPS than the new 3.18 version ]
see attached screenshot:  

Best Guess of Problem Origin:
-  missing CUPS components (known to be true)
-  missing or broken cups-filters or hplip components

Alexander, please let me know how I can help.

Dave

davygravy, 2019-04-15:
From CUPS error_log:

"Unable to execute ippfind utility: No such file or directory"
- driverless printing is pretty much broken/useless without the ippfind utility, so this binary ippfind is pretty essential

davygravy, 2019-04-15:
In cups-filters, reinstating:

`PKG_BUILD_DEPENDS:=libijs`

at line 22 and rebuilding the package seems to:
- make the package quite a bit larger (but now useful for users that have an inkjet (ubiquitous) printer
- increased the size of the package to comparable to a known working package that I'd built in 2017 for USB printer support

Next error is:
"No usable font available"

Is there a reason that the devavu-fonts-ttf was removed from the repo?

ryzhovau, 2019-04-15:
Thanks for feedback! I'll keep this issue open, as the only one witch printing related problems.

PDF and/or printing packages was heavily [reworked](https://t.me/Entware/3474) recently. Our CUPS was [replaced](https://github.com/Entware/entware-oldpackages-ports/commit/8e2e3afc28b1efb43aba98c79d97c68b5eeca6b9#diff-f39c3734919a8bd6e8214ba36422eb41) by [CUPS from upstream](https://github.com/openwrt/packages/tree/master/libs/libcups), but my answers below is about both new and old variants. In most cases, it was balance between monstrous dependencies and user comfort. Please, don't expect the same environment as in full-blown linux desktop, still, I'll be glad to find out how to improve printing packages without being overbloated.

* web interface was never fully functional on embedded devices, neither now, nor before. It uses some system utilities and perl/pyhon scripts which is often absent on NAS/routers.
* ippfind utility was never present on OpenWrt/Entware to avoid Avahi dependency. So, driver presumed to be configured manually.
* dejavu-fonts-ttf was never present in Entware, should we add it? I deleted obsolete liberation and ubuntu fonts recently, you can still install it from archive. I can't provide wide range of desired fonts, so, which one should I provide as essential one?
* How `libijs` enhances `cups-filters`? Is it really necessary?
* `ghostscript` is working, at least as PDF transformation tool, I never used it as a printing tool.

The most common CUPS mistakes is related to missing `/opt/...` paths from hard-coded binary parts and configs. Please, let me know where I can fix them too. 

Thanks again, Dave.

davygravy, 2019-04-15:
Spasibo, Alexander:
I'll work with it to see what minimal tweaks are needed to make it function.  I'll try to supply a patch set with a clear rationale for each patch.  This may take a week or so, as I work on it in my spare time.
Privyet,
Dave

davygravy, 2019-04-16:
Comparing what worked perfectly in CUPS 2.1.4:
```
D [11/Mar/2019:16:51:38 +0000] [Job 62] envp[22]=\"CONTENT_TYPE=application/vnd.cups-pdf-banner\"
D [11/Mar/2019:16:51:38 +0000] [Job 62] envp[23]=\"DEVICE_URI=socket://127.0.0.1:9100\"
D [11/Mar/2019:16:51:38 +0000] [Job 62] envp[24]=\"PRINTER_INFO=HPC4480p910nd printer on AirDisk\"
D [11/Mar/2019:16:51:38 +0000] [Job 62] envp[25]=\"PRINTER_LOCATION=BasementOffice\"
D [11/Mar/2019:16:51:38 +0000] [Job 62] envp[26]=\"PRINTER=HPC4480@AirDisk1:9100\"
D [11/Mar/2019:16:51:38 +0000] [Job 62] envp[27]=\"PRINTER_STATE_REASONS=none\"
D [11/Mar/2019:16:51:38 +0000] [Job 62] envp[28]=\"CUPS_FILETYPE=document\"
D [11/Mar/2019:16:51:38 +0000] [Job 62] envp[29]=\"FINAL_CONTENT_TYPE=application/vnd.cups-raster\"
D [11/Mar/2019:16:51:38 +0000] [Job 62] envp[30]=\"AUTH_INFO_REQUIRED=none\"
I [11/Mar/2019:16:51:39 +0000] [Job 62] Start rendering...
D [11/Mar/2019:16:51:39 +0000] cupsdMarkDirty(---J-)
D [11/Mar/2019:16:51:39 +0000] cupsdSetBusyState: newbusy="Dirty files", busy="Dirty files"
D [11/Mar/2019:16:51:39 +0000] [Job 62] Set job-printer-state-message to "Start rendering...", current level=INFO
D [11/Mar/2019:16:51:39 +0000] Discarding unused job-progress event...
D [11/Mar/2019:16:51:39 +0000] Discarding unused printer-state-changed event...
I [11/Mar/2019:16:51:39 +0000] [Job 62] Processing page 1...
D [11/Mar/2019:16:51:39 +0000] cupsdMarkDirty(---J-)
D [11/Mar/2019:16:51:39 +0000] cupsdSetBusyState: newbusy="Dirty files", busy="Dirty files"
D [11/Mar/2019:16:51:39 +0000] [Job 62] Set job-printer-state-message to "Processing page 1...", current level=INFO
D [11/Mar/2019:16:51:39 +0000] Discarding unused job-progress event...
D [11/Mar/2019:16:51:39 +0000] Discarding unused printer-state-changed event...
D [11/Mar/2019:16:51:40 +0000] [Client 1] GET /printers/HPC4480@AirDisk1:9100 HTTP/1.1
D [11/Mar/2019:16:51:40 +0000] cupsdSetBusyState: newbusy="Active clients and dirty files", busy="Dirty files"
D [11/Mar/2019:16:51:40 +0000] [Client 1] Read: status=200
D [11/Mar/2019:16:51:40 +0000] [Client 1] No authentication data provided.
D [11/Mar/2019:16:51:40 +0000] [Client 1] Processing GET /printers/HPC4480@AirDisk1:9100
D [11/Mar/2019:16:51:40 +0000] [CGI] argv[0] = "/opt/lib/cups/cgi-bin/printers.cgi"
D [11/Mar/2019:16:51:40 +0000] [CGI] envp[0] = "CUPS_CACHEDIR=/opt/var/cache/cups"
D [11/Mar/2019:16:51:40 +0000] [CGI] envp[1] = "CUPS_DATADIR=/opt/share/cups"
D [11/Mar/2019:16:51:40 +0000] [CGI] envp[2] = "CUPS_DOCROOT=/opt/share/doc/cups"
D [11/Mar/2019:16:51:40 +0000] [CGI] envp[3] = "CUPS_FONTPATH=/opt/share/cups/fonts"
D [11/Mar/2019:16:51:40 +0000] [CGI] envp[4] = "CUPS_REQUESTROOT=/opt/var/spool/cups"
D [11/Mar/2019:16:51:40 +0000] [CGI] envp[5] = "CUPS_SERVERBIN=/opt/lib/cups"

```

davygravy, 2019-04-16:
I believe I've worked around all the other problems, except this one above... Ghostscript is missing its CUPS device.  In the output below, you can see that it is missing... should the last item beginning with "c", before the "d"'s start...
```
GPL Ghostscript 9.26 (2018-11-20)
Copyright (C) 2018 Artifex Software, Inc.  All rights reserved.
Usage: gs [switches] [file1.ps file2.ps ...]
Most frequently used switches: (you can use # in place of =)
 -dNOPAUSE           no pause after page   | -q       `quiet', fewer messages
 -g<width>x<height>  page size in pixels   | -r<res>  pixels/inch resolution
 -sDEVICE=<devname>  select device         | -dBATCH  exit after last file
 -sOutputFile=<file> select output file: - for stdout, |command for pipe,
                                         embed %d or %ld for page #
Input formats: PostScript PostScriptLevel1 PostScriptLevel2 PostScriptLevel3 PDF
Default output device: bbox
Available devices:
   alc1900 alc2000 alc4000 alc4100 alc8500 alc8600 alc9100 ap3250 atx23
   atx24 atx38 bbox bit bitcmyk bitrgb bitrgbtags bj10e bj10v bj10vh bj200
   bjc600 bjc800 bjc880j bjccmyk bjccolor bjcgray bjcmono bmp16 bmp16m
   bmp256 bmp32b bmpgray bmpmono bmpsep1 bmpsep8 ccr cdeskjet cdj1600 cdj500
   cdj550 cdj670 cdj850 cdj880 cdj890 cdj970 cdjcolor cdjmono cdnj500 cfax
   chp2200 cif cljet5 cljet5c cljet5pr coslw2p coslwxl declj250 deskjet
   devicen dfaxhigh dfaxlow display dj505j djet500 djet500c dl2100 dnj650c
  ...
```

Anyone have an idea how to enable the CUPS device to be built?   I don't think that it is as simple as adding    --enable-cups   to the config.

ryzhovau, 2019-04-16:
Done. MIPSEL binaries arrived, the others will arrive later.
Which platform are you using?

davygravy, 2019-04-16:
armv7-2.6

Thank you, Alexander.  I will try this when I return home tonight.

davygravy, 2019-04-16:
Nice!  The patch was effective and my printer did spring to life, suddenly.  

gs -h shows the expected output:
```
gs -h | grep cups
   chp2200 cif cljet5 cljet5c cljet5pr coslw2p coslwxl cups declj250 deskjet
   /opt/share/cups/fonts/ : /opt/share/fonts/
```
In terms of function, I tested it with lp and printing a test page from CUPS... both worked fine.

I'll look now at some other items too, and look for brokenness or weird behavior.   I have been using a cups_2.2.10-2 that I rolled here, for some other reasons... but from here forward, I will revert to using just things from official Entware package feed (like the ghostscript that is now fixed) from here on out.

davygravy, 2019-04-17:
OK, testing CUPS, I see this:
```
root@AP1:/opt/share/cups/ipptool# ls
color.jpg                         get-subscriptions.test            onepage-letter.ps
create-printer-subscription.test  gray.jpg                          print-job-deflate.test
document-a4.pdf                   ipp-1.1.test                      print-job-gzip.test
document-a4.ps                    ipp-2.0.test                      print-job.test
document-letter.pdf               ipp-2.1.test                      testfile.jpg
document-letter.ps                ipp-2.2.test                      testfile.pcl
get-completed-jobs.test           ipp-everywhere.test               testfile.pdf
get-jobs.test                     onepage-a4.pdf                    testfile.ps
get-notifications.test            onepage-a4.ps                     testfile.txt
get-printer-attributes.test       onepage-letter.pdf                validate-job.test
root@AP1:/opt/share/cups/ipptool#  lpr -H 192.168.11.11 testfile.pdf 
lpr: Unsupported document-format "application/pdf".
root@AP1:/opt/share/cups/ipptool#  lpr -H 192.168.11.11 testfile.txt  
lpr: Unsupported document-format "text/plain".
root@AP1:/opt/share/cups/ipptool#  lpr -H 192.168.11.11 testfile.jpg 
lpr: Unsupported document-format "image/jpeg".
root@AP1:/opt/share/cups/ipptool#  lpr -H 192.168.11.11 testfile.ps
lpr: Unsupported document-format "application/postscript".
root@AP1:/opt/share/cups/ipptool#  lpr -H 192.168.11.11 testfile.pcl 
root@AP1:/opt/share/cups/ipptool# lpstat -t
scheduler is running
system default destination: HPC4480
device for HPC4480: socket://127.0.0.1:9100
HPC4480 accepting requests since Wed 17 Apr 2019 01:32:50 PM UTC
printer HPC4480 is idle.  enabled since Wed 17 Apr 2019 01:32:50 PM UTC
root@AP1:/opt/share/cups/ipptool# 
```

In the CUPS interface I see this:
```
E [17/Apr/2019:02:48:57 +0000] [Client 1] Returning IPP client-error-document-format-not-supported for Send-Document (ipp://localhost:631/printers/HPC4480) from localhost.
E [17/Apr/2019:13:31:00 +0000] [Client 4] Returning IPP client-error-document-format-not-supported for Send-Document (ipp://localhost:631/printers/HPC4480) from 192.168.11.11.
E [17/Apr/2019:13:32:18 +0000] [Client 5] Returning IPP client-error-document-format-not-supported for Send-Document (ipp://localhost:631/printers/HPC4480) from 192.168.11.11.
E [17/Apr/2019:13:32:26 +0000] [Client 6] Returning IPP client-error-document-format-not-supported for Send-Document (ipp://localhost:631/printers/HPC4480) from 192.168.11.11.
E [17/Apr/2019:13:32:41 +0000] [Client 7] Returning IPP client-error-document-format-not-supported for Send-Document (ipp://localhost:631/printers/HPC4480) from 192.168.11.11.
```

I'm not sure whether this is a problem with CUPS right now, or if it is something missing in HPLIP or CUPS-filters.  Testing of HPLIP function is limited because of the constrained nature of build [hp-check doesn't function], but I should be able to track down the error.

davygravy, 2019-04-17:
1.  cups-filters is missing some necessary components in /opt/share.  Without these, USB printing fails on many printers.

```
diff -aur ./a/Makefile ./b/Makefile
--- ./a/Makefile	2019-04-17 11:08:22.922534934 -0500
+++ ./b/Makefile	2019-04-17 10:26:12.537987000 -0500
@@ -11,7 +11,7 @@
 
 PKG_NAME:=cups-filters
 PKG_VERSION:=1.22.1
-PKG_RELEASE:=1
+PKG_RELEASE:=2
 
 PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.bz2
 PKG_SOURCE_URL:=https://github.com/OpenPrinting/cups-filters/releases/download/release-1-22-1
@@ -75,6 +75,8 @@
 	$(CP) $(PKG_INSTALL_DIR)/opt/lib/cups $(1)/opt/lib
 	$(INSTALL_DIR) $(1)/opt/sbin
 	$(INSTALL_BIN) $(PKG_INSTALL_DIR)/opt/sbin/cups-browsed $(1)/opt/sbin
+	$(INSTALL_DIR) $(1)/opt/share
+	$(CP) $(PKG_INSTALL_DIR)/opt/share/* $(1)/opt/share
 endef
 
 $(eval $(call BuildPackage,cups-filters))
```

2.  Printing using `lp` is failing for me unless I install at least one font.  The liberation font package in archive is sufficient.

3.  Missing in libcups:  3 libs

libcupsmime
libcupsppdc
libcupscgi

are not present in /opt/lib/

These three libs -are- included in the old packaging system (archives), as three separate packages.  I think its a good idea to roll them into a single package, but perhaps something in the build process (stripping not happening correctly?) is going on.

Oddly enough 3 distinct binaries (one in each of the mime, ppdc and cgi dirs, IIRC) are --larger-- than they should be, by almost the exact size of the libs that they are associated with.

davygravy, 2019-04-23:
Great! & thanks...

I will test this all over from scratch using only vanilla ... on my extra router ASAP and report results back.    Will let you know if everything works as expected.

klausrollmann, 2019-04-25:
I have the exact same problem. I can help testing if you want.

The fix is simply to build cups-filter and ghostscript from git and download the fonts and libcups* packages from archive? Or am I missing something?

ryzhovau, 2019-04-25:
> The fix is simply to build cups-filter and ghostscript from git

…to build it with some options. This part already done.

As for fonts, I'll add [dejavu-fonts-ttf](https://github.com/openwrt/packages/blob/master/fonts/dejavu-fonts-ttf/Makefile) as OpenWrt guys do.

davygravy, 2019-04-25:
With what I see this looks good in terms of function ... here is how I tested it...
Using a fresh install of Entware on armv7sf-k2.6, this is how I tested it:

    opkg install cups-filters ghostscript hplip-full libcups cups dejavu-fonts-ttf-DejaVuSansMono   # CUPS Printing Stack
    nano /opt/etc/cups/cups-files.conf     # see the link below for primer on CUPS server configuration
    nano /opt/etc/cups/cupsd.conf
    cupsd -t            # to check the config files
    /opt/etc/init.d/S60cupsd start

The server/scheduler for CUPS runs as expected, and I accessed the web interface through port 631.  I added my USB printer and was able to print "Test Page" via the webif, which is very nice.

Also tested it with 

     echo  testing  |  lp
which also worked as expected.

@ryzhovau , nicely done, this looks to be Okey-Dokey in terms of working.

My only two questions are sort of 'cosmetic':
1.  I see that when installing the "CUPS Printing Stack" above --without-- the dejavu-font-ttf-DejaVuSansMono package, lp complained to me, saying: 

      "No usable font available"
in the webif -> Jobs, and in /opt/var/log/cups/Error_log there are mentions like:

         W [25/Apr/2019:17:24:11 +0000] [Job 2] Ignored bad font \"monospace\"
         W [25/Apr/2019:17:24:11 +0000] [Job 2] Ignored bad font \"monospace\"
         E [25/Apr/2019:17:24:11 +0000] [Job 2] No usable font available
         ...
         D [25/Apr/2019:17:24:14 +0000] [Job 2] No viable font found
         D [25/Apr/2019:17:24:14 +0000] [Job 2] Set job-printer-state-message to "Ignored bad font \"monospace\"", current level=WARN
         D [25/Apr/2019:17:24:14 +0000] [Job 2] No viable font found
         D [25/Apr/2019:17:24:14 +0000] [Job 2] Set job-printer-state-message to "No usable font available", current level=ERROR

Once I explicitly installed dejavu-fonts-ttf-DejaVuSansMono via 'opkg' , everything seemed very happy and no further errors were generated.

It is my understanding that a font package such as dejavu-fonts-ttf-DejaVuSansMono is only needed if the user wishes to print from the CLI using 'lp'.

NOTE to anyone reading this comment:  a nice primer on CUPS cupsd.conf setup is here:    http://chschneider.eu/linux/server/cups.shtml

2.  Were you intending that the package libcups would have all of the libcups* libraries in the package?  There are only 2 inside (libcups, libcupsimage) and I think 3 libraries are either missing (libcupsppdc, libcupsmime, libcupscgi, or are "wrapped" with some binaries involving those functions.

klausrollmann, 2019-04-25:
I am using MIPSEL Kernel 2.6.22 and I just found out that entware dropped support to mips kernel < 3.4. When did that happen? Is there any way I ccould use these packages for mips kernel 2.6?

davygravy, 2019-04-25:
@klausrollmann  :  It is my understanding that  maurer  @ linksysinfo site ( reference : https://github.com/maurerr/Entware-backports   and   https://www.linksysinfo.org/index.php?threads/mips-repo-entware-backports-entware-ng-reloaded.74337/#post-304426 )  is attempting to address this by backporting some packages ...

You may be able to clone his repo, attempt to refresh the CUPS and print stack ... and bring this to mipselsf-k2.6 ...   I'm not sure how much infrastructure (base or dependent packages) would have to be upgraded...

ryzhovau, 2019-04-26:
> I see that when installing the "CUPS Printing Stack" above --without-- the dejavu-font-ttf-DejaVuSansMono package, lp complained to me, saying:
"No usable font available"

`cups-filters` uses some [hard-coded](https://github.com/Entware/rtndev/commit/bdc3fd15d8157d8604b0daf83f1671764c928035#diff-96cec8527268f6bf1af2c910e5408bb6R49) default font, so I pulled it as as [dependency](https://github.com/Entware/rtndev/commit/bdc3fd15d8157d8604b0daf83f1671764c928035#diff-96cec8527268f6bf1af2c910e5408bb6R29). Looks like I should change it to `dejavu-fonts-ttf-DejaVuSansMono`.

> Were you intending that the package libcups would have all of the libcups* libraries in the package? There are only 2 inside (libcups, libcupsimage) and I think 3 libraries are either missing (libcupsppdc, libcupsmime, libcupscgi, or are "wrapped" with some binaries involving those functions.

I'm still not sure those 3 additional libraries is necessary. We trying to keep `libcups` dependencies on minimal level. I prefer to keep it current state if it's working as expected.

Thank you for helping me to keep printing stuff is in consistent state.

closed 2019-04-26 by ryzhovau

davygravy, 2019-04-26:
Happy to help!   Again, thanks to you, zyxmon and others for continuing this project.

davygravy, 2019-04-27:
> I am using MIPSEL Kernel 2.6.22 and I just found out that entware dropped support to mips kernel < 3.4. When did that happen? Is there any way I ccould use these packages for mips kernel 2.6?

@klausrollmann ,  this issue with Entware-NG (including mipselsf-k2.6) is solved now here :  https://github.com/davygravy/AirPrint-on-Entware/wiki/AirPrint-emulation-on-Entware-with-fewer-hiccups-:-HOWTO

klausrollmann, 2019-04-27:
@davygravy Thanks for adding support to mipselsf-k2.6.

I followed the tutorial, but still get "Filter failed" when I try to print a test page on cups.

davygravy, 2019-04-27:
> > Were you intending that the package libcups would have all of the libcups* libraries in the package? There are only 2 inside (libcups, libcupsimage) and I think 3 libraries are either missing (libcupsppdc, libcupsmime, libcupscgi, or are "wrapped" with some binaries involving those functions.
> 
> I'm still not sure those 3 additional libraries is necessary. We trying to keep `libcups` dependencies on minimal level. I prefer to keep it in current state if it's working as expected.
> 

A friendly   FYI @ryzhovau ...  just in case you ever run into a problem looking for these...
I found the reason that those 3 libraries are missing:  a change in  /cgi-bin/Makefile,  
 ppdc/Makefile  and  scheduler/Makefile

The reference/diff   is here : https://fossies.org/diffs/cups/2.2.8-source_vs_2.2.9-source/     in the Makefile section/table.

Now it is clear to me why libraries are no longer generated, apparently by design.  ;)

davygravy, 2019-04-27:
> @davygravy Thanks for adding support to mipselsf-k2.6.
> 
> I followed the tutorial, but still get "Filter failed" when I try to print a test page on cups

@klausrollmann :  please open an issue at my github repo, so that alex and zyxmon aren't getting spammed by me and you. :)  I'll try to help  you there.  :)