Entware/Entware #271: Possible recursive or improperly nested directories in /opt

davygravy, 2019-04-17:
On my installation of Entware on a Fresh Tomato (19.1 firmware) router, I see that there looks to be something out of location, perhaps:

Directory structure:
/opt/tmp/opt/lib/opkg/

Is this intended to be :?
/opt/lib/opkg/

I thought that maybe there was some unexpected symlinking going on... but I can't seem to find it...

```ls -lna /opt/tmp/opt/lib/opkg/
drwxr-xr-x    3 0        0             4096 Apr 16 19:17 .
drwxr-xr-x    3 0        0             4096 Apr 16 19:16 ..
drwxr-xr-x    2 0        0             4096 Apr 16 19:16 info
-rw-r--r--    1 0        0                0 Apr 16 21:13 status

```

As far as I can tell, it does not cause me an error when using opkg, but it might if a config is changed... or if someone was trying to build something and didn't expect this structure...

zyxmon, 2019-04-17:
`/opt/tmp/opt/lib/opkg....` is used internally and should contain zero length files or empty directories.

davygravy, 2019-04-17:
```
root@AP1:/opt/share/cups/ipptool# cat /opt/etc/opkg.conf 
src/gz entware http://bin.entware.net/armv7sf-k2.6
dest root /
dest ram /opt/tmp
lists_dir ext /opt/var/opkg-lists
option tmp_dir /opt/tmp
arch all 100
arch armv7soft 150
arch armv7-2.6 160
```

zyxmon, 2019-04-17:
This is normal if status length is zero.

davygravy, 2019-04-18:
OK, thank you.  I'm closing it since it seems like a non-issue.

closed 2019-04-18 by davygravy