Entware/Entware #274: BIND 9.12.3-P4 - TCP_FASTOPEN fail

protogen, 2019-04-20:
Over the last two days I completely rebuilt my RT-AC88U on Merlin 384.10_2.  I did a factory reset and reformatted both JFFS and USB before reinstalling Entware (64bit).

Upon starting BIND 9.12.3-P4 I saw the following error in my logs.
```
named[2126]: [daemon.info]  listening on IPv6 interfaces, port 53
named[2126]: [daemon.err]  socket.c:5683: unexpected error:
named[2126]: [daemon.err]  setsockopt(21, TCP_FASTOPEN) failed with Protocol not available
named[2126]: [daemon.info]  listening on IPv4 interface lo, 127.0.0.1#53
named[2126]: [daemon.err]  socket.c:5683: unexpected error:
named[2126]: [daemon.err]  setsockopt(22, TCP_FASTOPEN) failed with Protocol not available
named[2126]: [daemon.info]  listening on IPv4 interface eth0, <my external IP>#53
named[2126]: [daemon.err]  socket.c:5683: unexpected error:
named[2126]: [daemon.err]  setsockopt(23, TCP_FASTOPEN) failed with Protocol not available
named[2126]: [daemon.info]  listening on IPv4 interface br0, 192.168.9.1#53
named[2126]: [daemon.err]  socket.c:5683: unexpected error:
named[2126]: [daemon.err]  setsockopt(24, TCP_FASTOPEN) failed with Protocol not available
...
named[2126]: [daemon.err]  socket.c:5683: unexpected error:
named[2126]: [daemon.err]  setsockopt(25, TCP_FASTOPEN) failed with Protocol not available
named[2126]: [daemon.notice]  command channel listening on 127.0.0.1#953
```

The BIND server process (named) was running, but the output of `netstat -anp` showed it was not listening on any interface.  DNS resolution was not working at all.

To get around this problem, I ended up staying on 384.10_2 and created a modified `entware-setup.sh` which used the old 32bit repository (http://pkg.entware.net/binaries/armv7).  I then re-installed Entware from scratch, including BIND 9.11.2-3 (32bit).  This version of BIND started without error and is running perfectly.

This result is confusing because BIND has used TCP_FASTOPEN since version 9.11.0 - so why does 9.11.2-3 (32bit) work but 9.12.3-P4 (64bit) fail?  Is it something to do with 32bit vs 64bit?

From my research, it appears that for TCP_FASTOPEN to work it requires kernel support.  I'm unclear whether this means there is a problem with how kernel 384.10_2 has been compiled, or there's a problem with how BIND 9.12.3-P4 (64bit) has been compiled.

Or is it a problem with one of the Entware 64bit libraries?

dnmTX, 2019-04-20:
@protogen i'm not an expert but in case the problem is not in the x64bit version you can at least check on that build that you have problems with if tcp_fastopen is enabled in the kernel.From shell:
`sysctl -a | grep fastopen`
see if that string exist and if it does what is set to.The ideal is 3.

protogen, 2019-04-20:
@dnmTX Thanks for your response.

I made a mistake regarding 32bit vs 64bit.  Everything is 32bit on my router model (nothing 64bit).  I've edited my original post.

That said, I've checked my router and unfortunately I don't have the `sysctl` binary.
```
[605] admin@router:~ # which sysctl
[606] admin@router:~ # find /bin | grep -i sysctl
[607] admin@router:~ # opkg search *sysctl
[608] admin@router:~ #
```
I've also checked `/proc` but I don't see any `tcp_fastopen`.

```
[614] admin@router:~ # find /proc | grep -i tcp_fastopen
[615] admin@router:~ #
```

dnmTX, 2019-04-20:
@protogen run the command that i posted above just in case.This will show you for sure if you have it or not.

protogen, 2019-04-20:
@dnmTX I did that first.  The command doesn't exist.

```
[616] admin@router:~ # sysctl
bash: sysctl: command not found
```

dnmTX, 2019-04-20:
@protogen yeah...sorry(read my edit above).Here is some info also:
https://www.snbforums.com/threads/editing-nvram-variables.40651/#post-341693

protogen, 2019-04-20:
@dnmTX Hehe... I read the same thread too.  Thanks for replying, it's appreciated.

dnmTX, 2019-04-20:
@protogen just another wild guess as i don't use BIND but just in case check in bind's options if there is something to set it not even to use tcp_fastopen. Again,guessing here but it's worth the try and GOOD LUCK 👍

protogen, 2019-04-21:
@dnmTX TCP_FASTOPEN is not a user configurable item in BIND.  In fact, I don't believe it's a user configurable item in any binary.  It's a compile-time choice.

The cause of this problem is probably due to how either Merlin kernel 384.10_2 was compiled , or how BIND 9.12.3-P4 was compiled. I'm just not sure which.  This probably needs comment from the Entware contributors.

protogen, 2019-04-21:
I received a response from Merlin on the SNBForums.  To quote...

"TCP Fast Open requires Linux 3.7. Your router runs 2.6.36, so Fast Open was never supported on that model."

Given my router does not support TCP Fast Open why is BIND 9.12.3-P4 trying to use it?

zyxmon, 2019-04-21:
>> Given my router kernel does not support TCP Fast Open why is BIND 9.12.3-P4 trying to use it?
Because it is defined at compile time, not at run time.
Please specify your Entware feed (architecture) not your router model.

protogen, 2019-04-21:
```
[765] admin@router:~ # uname -a
Linux router 2.6.36.4brcmarm #1 SMP PREEMPT Wed Apr 3 18:39:55 EDT 2019 armv7l ASUSWRT-Merlin
```

zyxmon, 2019-04-21:
This is your router architecture. Iэму asked you about Entware feed (architecture). I guess it is armv7sf-k2.6. I'll probably make a fix for 2.6.36 and BIND today.

protogen, 2019-04-21:
> This is your router architecture. I've asked you about Entware feed (architecture).

Sorry.  I'm a old Unix/Linux sys admin.  I hear 'architecture' and immediately go for the uname command.  I've edited my comment above with the _correct_ info you requested.

> I'll probably make a fix for 2.6.36 and BIND today.

Thank you very much.

zyxmon, 2019-04-21:
Please update your bind packages. The are not using TCP_FASTOPEN now.

protogen, 2019-04-22:
BIND 9.12.3-P4-1a started without error and is running perfectly.  Problem solved.

Thank you for fixing this so quickly.
```
named[1187]: [daemon.notice]  starting BIND 9.12.3-P4 <id:a3d2ae0>
...
named[1187]: [daemon.info]  listening on IPv4 interface lo, 127.0.0.1#53
named[1187]: [daemon.info]  listening on IPv4 interface eth0, <my external IP>#53
named[1187]: [daemon.info]  listening on IPv4 interface br0, 192.168.9.1#53
...
named[1187]: [daemon.notice]  command channel listening on 127.0.0.1#953
```
```
[504] admin@router:~ # netstat -anp | grep ':53 '
tcp        0      0 192.168.9.1:53          0.0.0.0:*               LISTEN      1187/named
tcp        0      0 <my external IP>:53     0.0.0.0:*               LISTEN      1187/named
tcp        0      0 127.0.0.1:53            0.0.0.0:*               LISTEN      1187/named
udp        0      0 192.168.9.1:53          0.0.0.0:*                           1187/named
udp        0      0 <my external IP>:53     0.0.0.0:*                           1187/named
udp        0      0 127.0.0.1:53            0.0.0.0:*                           1187/named
```

closed 2019-04-22 by protogen