Entware/Entware #275: Package request: python 3.7

amurabito, 2019-04-27:
Package request
---
Update to python 3 installations to latest versions. 

- [ python 3.6 is installed but could be upgraded ] 
- [ X ] Package is in active development (at least, not abandoned for 5 years or more),
- [ X ] I'm ready to test added package on my {platform} and get a feedback about it.

---
Please, fill `{}` fields when asking for new package or delete this template in any other case.

ryzhovau, 2019-04-28:
Package will be synced with [upstream](https://github.com/openwrt/packages/blob/master/lang/python/python3-version.mk) at the (approx.) last May decade.

ryzhovau, 2019-05-08:
Looks like you are on Entware-ng or some other obsolete fork.

Python 3.7 [already there](https://bin.entware.net/x64-k3.2/Packages.html).

closed 2019-05-08 by ryzhovau