Entware/Entware #276: nzbget: error while loading shared libraries after opkg upgrade

amplatfus, 2019-04-29:
Hi,

I just did opkg upgrade and afterwords nzbget failed to start because of libssl.so.1.0.0 missing. Could you please give a hint how to fix it?

Thank you so much!
```

root@rooter:/jffs# opkg update
Downloading http://bin.entware.net/armv7sf-k2.6/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/entware
root@rooter:/jffs# opkg upgrade
Upgrading libopenssl on root from 1.0.2p-1a to 1.1.1a-2...
Downloading http://bin.entware.net/armv7sf-k2.6/libopenssl_1.1.1a-2_armv7-2.6.ipk
Removing obsolete file /opt/lib/libssl.so.1.0.0.
Removing obsolete file /opt/lib/libcrypto.so.1.0.0.

root@rooter:/tmp/home/root# nzbget -s
nzbget: error while loading shared libraries: libssl.so.1.0.0: cannot open shared object file: No such file or directory

```

ryzhovau, 2019-04-29:
https://github.com/Entware/Entware/wiki#note-about-packages-upgrade

closed 2019-04-29 by ryzhovau

amplatfus, 2019-04-29:
Hi,

In the meantime I fixed using:

```
opkg --autoremove remove nzbget
opkg install nzbget
```

Thank you!
amplatfus