Entware/Entware #277: mips - generic.sh: line 43: /opt/bin/opkg: Permission denied

viotemp1, 2019-04-30:
Package request
---

Hello, 
I tried to install on my dd-wrt and I receive Permission denied error. What should I do?
These are the steps and some logs:

Firmware: DD-WRT v3.0-r37305 std (10/10/18)
Router Name  DD-WRT
Router Model  Dlink DIR-882 A1
Firmware Version DD-WRT v3.0-r37305 std (10/10/18)
Kernel VersionLinux 3.10.108-d6 #1741 SMP Wed Oct 10 16:21:20 CEST 2018 mips

uname -a
Linux mydlinkrouter 3.10.108-d6 #1741 SMP Wed Oct 10 16:21:20 CEST 2018 mips DD-WRT

mount
rootfs on / type rootfs (rw)
/dev/root on / type squashfs (ro,relatime)
proc on /proc type proc (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
ramfs on /tmp type ramfs (rw,relatime)
none on /dev type tmpfs (rw,relatime,size=512k)
devpts on /dev/pts type devpts (rw,relatime,mode=600)
devpts on /proc/bus/usb type usbfs (rw,relatime)
/dev/sda1 on /opt type ext4 (rw,relatime,data=ordered)
/dev/sda1 on /tmp/mnt/sda1 type ext4 (rw,relatime,data=ordered)

/opt# wget http://bin.entware.net/mipssf-k3.4/installer/generic.sh
Connecting to bin.entware.net (104.27.176.50:80)

/opt# sh generic.sh
Info: Checking for prerequisites and creating folders...
Warning: Folder /opt exists!
Warning: Folder /opt/bin exists!
Warning: If something goes wrong please clean /opt folder and try again.
Warning: Folder /opt/etc exists!
Warning: If something goes wrong please clean /opt folder and try again.
Warning: Folder /opt/lib/opkg exists!
Warning: If something goes wrong please clean /opt folder and try again.
Warning: Folder /opt/tmp exists!
Warning: If something goes wrong please clean /opt folder and try again.
Warning: Folder /opt/var/lock exists!
Warning: If something goes wrong please clean /opt folder and try again.
Info: Opkg package manager deployment...
Connecting to bin.entware.net (104.27.177.50:80)
Connecting to bin.entware.net (104.27.176.50:80)
Connecting to bin.entware.net (104.27.177.50:80)
Connecting to bin.entware.net (104.27.176.50:80)
Connecting to bin.entware.net (104.27.177.50:80)
Connecting to bin.entware.net (104.27.176.50:80)
Info: Basic packages installation...
generic.sh: line 43: /opt/bin/opkg: Permission denied
generic.sh: line 44: /opt/bin/opkg: Permission denied
Info: Congratulations!
Info: If there are no errors above then Entware was successfully initialized.
Info: Add /opt/bin & /opt/sbin to your PATH variable
Info: Add '/opt/etc/init.d/rc.unslung start' to startup script for Entware services to start
Info: Found a Bug? Please report at https://github.com/Entware/Entware/issues

- [ ] No similar package already added and/or this one got some new killer feature(s),
- [ ] Package is in active development (at least, not abandoned for 5 years or more),
- [ ] I'm ready to test added package on my {platform} and get a feedback about it.

---
Please, fill `{}` fields when asking for new package or delete this template in any other case.

zyxmon, 2019-04-30:
Your devices is mipsel, not mips!

closed 2019-04-30 by zyxmon

viotemp1, 2019-04-30:
sorry, my bad. You are perfectly right!
All ok now: opkg --version
opkg version d4ba162bb3f931bd5a019154270a548cf1f0853b (2019-01-31)