Entware/Entware #279: Request package: ttyload

jaalto, 2019-05-01:
Package request
---
https://packages.debian.org/testing/ttyload

Console based color-coded graphs of CPU load average

Display fairly standard, but hard-coded, ANSI ASCII escape sequences for screen manipulation and colorization for different graphs: 1 minute, 5 minute, 15 minute load averages.

- [ x] No similar package already added and/or this one got some new killer feature(s),
- [ x] MATURE/STABLE
- [x] I'm ready to test added package on my {platform} and get a feedback about it.

---
Please, fill `{}` fields when asking for new package or delete this template in any other case.

ryzhovau, 2019-05-03:
We've got `htop`, `atop` and standard `top` utility, showing system load avg over the last 1, 5 and 15 minutes. Why we should add another tool, abandoned 5 years ago?

jaalto, 2019-05-03:
(1) abandoned is not a criteria The tool is mature, never had had any problems
(2) no similar. TTYLOAD will display ASCII graphics of the load balance when run. Very useful without GUI

ryzhovau, 2019-05-06:
Can't say ASCII art is a major feature of `top`-like utility. 
Won't add it, sorry.

closed 2019-05-06 by ryzhovau