Entware/Entware #283: openssh fails with "No user exists for uid 0"

saigelosli, 2019-05-05:
Router model: Netgear R7000
Firmware: Advanced Tomato v3.5-140 AIO-64k
Entware version:
  # opkg --version
  opkg version d4ba162bb3f931bd5a019154270a548cf1f0853b (2019-01-31)

I had a previous version of Entware installed, but wanted access to some updates. So I wiped out the existing installation with:
  # rm -r /opt/*
Then I re-installed Entware with:
  # entware-install.sh
Updated the package database (twice, for good measure):
  # opkg-update
  # opkg-upgrade
  # opkg-update
  # opkg-upgrade
Re-installed the packages I had before (abbreviated list shown here, otherwise too long):
  # opkg install alsa-lib bind-check bind-client ...
Package installation seemed to go smoothly. However, when I run the "ssh" command, I get a new error message I've never seen before, and it won't connect anywhere:
  # which ssh
  /opt/bin/ssh
  # ssh user@host
  No user exists for uid 0

zyxmon, 2019-05-05:
We have no installation script called entware-install.sh. See - https://github.com/Entware/Entware/wiki/Alternative-install-vs-standard
We cannot help you - we do not know what you have installed. Ask script manufacturer for support.

saigelosli, 2019-05-05:
You're right! Thank you for suggesting to check [https://github.com/Entware/Entware/wiki/Alternative-install-vs-standard](https://github.com/Entware/Entware/wiki/Alternative-install-vs-standard), that took me to the right place to find your up-to-date "generic.sh" installer, which I used to re-install Entware. Now it works fine.

closed 2019-05-05 by saigelosli