Entware/Entware #292: Request to add openvpn package 2.4.6-3 WITH plugin support

jberlet, 2019-05-20:
Can we add a openvpn-openssl package (openvpn 2.4.6-3) with a compile option of 'enable_plugins' = yes option? This option is enabled by default in the default openvpn server build config. so am not sure why this continues to be a deprecated openvpn compilation feature in the current Entware openvpn-openssl opkg as well as the default dd-wrt package.

Enabling this feature would allow use of thirdparty MultiFactor Authentication like Duo and other cool plugins with OpenVPN. Thank you!

JB
---

jberlet, 2019-06-04:
BTW - I customized the Entware openvpn package to build/enable the openvpn_pam_authentication plugin (.so). It would be nice to have this included in future entware openvpn builds by default (with enable_plugins turned on by default).
JB

ryzhovau, 2019-10-10:
This package came from [upstream](https://github.com/openwrt/openwrt/blob/master/package/network/services/openvpn/Makefile#L83) with no changes. 

I was keeping this issue open for 3 months and see no interest from other users. So I'm closing it for now. 
Sorry JB.

Brycey92, 2024-05-13:
Hi, is there any chance of this happening now that it's 5 years later? The lack of plugin support is limiting what I can do with my NAS.