Entware/Entware #296: Request package - borg backup utility

Evidlo, 2019-05-31:
Package request
---
[borg](https://github.com/borgbackup/borg/) is a CLI backup utility.  It supports snapshots and automatically de-duplicates data.  There are ARM builds already available [here](https://borg.bauerj.eu/).

- [x] No similar package already added and/or this one got some new killer feature(s),
- [x] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I'm ready to test added package on my {platform} and get a feedback about it.

---

bugsyb, 2019-10-26:
Signing up under the request.
I've tried myself to use pip3 and after going through number of hops got as far as:
[code]
    aarch64-openwrt-linux-gnu-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -O2 -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -O2 -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -DNDEBUG -fno-inline -O2 -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -DNDEBUG -fno-inline -I/media/ware/Entware.2019.10/staging_dir/target-aarch64_cortex-a53_glibc-2.27/opt/include -I/media/ware/Entware.2019.10/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.4.0_glibc-2.27/include -fPIC -D__LITTLE_ENDIAN__=1 -I/opt/local/include -Isrc/borg/algorithms/lz4/lib -Isrc/borg/algorithms/zstd/lib -Isrc/borg/algorithms/zstd/lib/common -Isrc/borg/algorithms/zstd/lib/compress -Isrc/borg/algorithms/zstd/lib/decompress -Isrc/borg/algorithms/zstd/lib/dictBuilder -Isrc/borg/algorithms/blake2/ref -I/opt/include/python3.7 -c src/borg/algorithms/msgpack/_packer.cpp -o build/temp.linux-aarch64-3.7/src/borg/algorithms/msgpack/_packer.o
    src/borg/algorithms/msgpack/_packer.cpp:4:10: fatal error: Python.h: No such file or directory
     #include "Python.h"
              ^~~~~~~~~~
    compilation terminated.
    error: command 'aarch64-openwrt-linux-gnu-gcc' failed with exit status 1
[/code]

Trying to build statically linked binary under Ubuntu on same system (in fact second OS image) didn't succeed.

BorgBackup offers Armv8 binaries but these result in:
[code]
./borg-1.1.10-arm64
/bin/sh: ./borg-1.1.10-arm64: not found
[/code]

For some of knowledgeable guys it won't be a problem to go over this.

Hopefully a good soul will look into it.

p.s. memory is not an issue, as i.e. in my case it is Odroid N2 with 4GB RAM.

Evidlo, 2019-10-30:
@bugsyb Whenever you see 'Python.h: No such file or directory' it means you are missing `python-dev` or `python3-dev`.

bugsyb, 2020-01-08:
@Evidlo, thanks. The ask is to have package provided as part of the Entware even if I've tried to compile it myself.

K3rm1t4, 2020-01-26:
I second this request.

closed 2020-01-30 by The-BB

bugsyb, 2020-02-03:
@The-BB - thank you!

bugsyb, 2020-02-03:
Just tried and something is missing :( 
```
### install
# borg --version
Traceback (most recent call last):
  File "/opt/bin/borg", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/__init__.py", line 3191, in <module>
  File "/__init__.py", line 3175, in _call_aside
  File "/__init__.py", line 3204, in _initialize_master_working_set
  File "/__init__.py", line 583, in _build_master
  File "/__init__.py", line 900, in require
  File "/__init__.py", line 786, in resolve
pkg_resources.DistributionNotFound: The 'borgbackup==1.1.10' distribution was not found and is required by the application

### trying to run
# borg --version
Traceback (most recent call last):
  File "/opt/bin/borg", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/__init__.py", line 3191, in <module>
  File "/__init__.py", line 3175, in _call_aside
  File "/__init__.py", line 3204, in _initialize_master_working_set
  File "/__init__.py", line 583, in _build_master
  File "/__init__.py", line 900, in require
  File "/__init__.py", line 786, in resolve
pkg_resources.DistributionNotFound: The 'borgbackup==1.1.10' distribution was not found and is required by the application

#
```

The-BB, 2020-02-03:
This is due to a missing dependency `python3-pkg-resources`. Please install it manually `opkg install python3-pkg-resources`

bugsyb, 2020-02-03:
Something's missing still and says the same "DistributionNotFound: The 'borgbackup==1.1.10'"

Did run below one after another.
```
# opkg install python3-pkg-resources
Installing python3-pkg-resources (41.2.0-1) to root...
Downloading http://bin.entware.net/aarch64-k3.10/python3-pkg-resources_41.2.0-1_aarch64-3.10.ipk
Configuring python3-pkg-resources.

# borg
Traceback (most recent call last):
  File "/opt/bin/borg", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/__init__.py", line 3191, in <module>
  File "/__init__.py", line 3175, in _call_aside
  File "/__init__.py", line 3204, in _initialize_master_working_set
  File "/__init__.py", line 583, in _build_master
  File "/__init__.py", line 900, in require
  File "/__init__.py", line 786, in resolve
pkg_resources.DistributionNotFound: The 'borgbackup==1.1.10' distribution was not found and is required by the application

```

bugsyb, 2020-02-05:
Just to close subject - updated packages (`opkg update`) and borg re-installed, it works now.