Entware/Entware #297: Problem with libcurl compiled with mbedtls

savonarola27, 2019-05-31:
I have a problem running pyload on my QNAP TS-119 (ARM5 - fw. 4.3.3.) and QNAP TS-231P (ARM7 - fw. 4.3.6).

after some investigation I found that perhaps the problem is due to the libcurl compiled with opensbed's mbedtls invce.

Below one of the many tests that returns the error in question:

Steps:
- manual installation Entware package Entware_1.00std.qpkg via App Center
- opkg update
- opkg upgrade

```
[~] # /opt/bin/python2.7
Python 2.7.16 (default, May 23 2019, 05:34:26)
[GCC 7.4.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import pycurl
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: pycurl: libcurl link-time ssl backend (openssl) is different from compile-time ssl backend (mbedtls)
>>> pycurl.version
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'pycurl' is not defined
>>>

```

How can I have lib curl installed with openssl?

zyxmon, 2019-05-31:
I do confirm the bug for all architectures
python-curl is build with mbedtls as was specified by openwrt Makefile. 
There was a fix recently - https://github.com/openwrt/packages/commit/0dd445a248c629b37923ae88999abd2fb19ec1a5#diff-2b5a1960315332f629c18bb3507b1773
We will try to use it and report here if it helps

savonarola27, 2019-05-31:
Great! I wil follow this post for any news.

Thank you very much!

savonarola27, 2019-05-31:
Meanwhile can i downgrade to old versions of python-curl and libcurl packages? If so, which versions?

zyxmon, 2019-05-31:
Jusn run `opkg updatel opkg upgrade`. The issue is fixed.

savonarola27, 2019-05-31:
Yeah! Now it works!!! Also pyload! 

Thank you very much.

closed 2019-05-31 by zyxmon