Entware/Entware #298: unfs3 doesn't auto start

Magissia, 2019-06-02:
unfs3 doesn't auto start, initrd file was left as is and is executable, if starting it manually using /opt/etc/init.d/S56unfsd start, it starts properly and nfs shares are accessible.

0.9.23-20150511-1 is installed, there are sadly no newer version in the repositories.

ryzhovau, 2019-06-03:
Did you mean `unfs3` won't start at boot time? 
Please, make sure necessary network network connection ready at this moment.

You may add `/opt/sbin/unfsd` manually somewhere to catch error message.

Magissia, 2019-06-03:
Yes, at boot time.
I am able to SSH in before any init.d scripts are started, does it mean necessary network connection is (are) ready at this moment ?

Should /opt/sbin/unfsd be added somewhere in an init.d script other than the unfs3 script ?

zyxmon, 2019-06-04:
>> does it mean necessary network connection is (are) ready at this moment ?

It means lan is up. It does not meen wan is up. Try to add sleep command to S56unfsd to check if script tries to start unf3 too early.

closed 2019-06-13 by ryzhovau

Magissia, 2019-07-01:
Adding `sleep 1m` did resolve the problem, but I don't understand why WAN would be required for this LAN service ?