Entware/Entware #302: sshpass 1.06

tegelad, 2019-06-06:
Package request
---
The ability to use the NAS device as a jumpbox and not having to enter a password.
I have a situation where trusted keys are "verboten", and I need to use password authentication (due to GSS-API).

When I attempted to compile the version from git hub it bails on the awk creation of subs.awk in the most horrible and evil ways ....

As I don't spin up things on debian or arm often, I am hoping someone here has the ability to dive in and update it ....

:) and the benefits ... well you can imagine the fun of not having to type the password in .....

- [X] No similar package already added and/or this one got some new killer feature(s),
- [ ] Package is in active development (at least, not abandoned for 5 years or more),
- [ ] I'm ready to test added package on my {platform} and get a feedback about it.

---
Please, fill `{}` fields when asking for new package or delete this template in any other case.

ryzhovau, 2019-06-07:
Its a bad practice to put passwords into scripts, please use key authentication.
Won't add it, sorry.

https://github.com/Entware/Entware-ng/issues/26

closed 2019-06-07 by ryzhovau

tegelad, 2019-06-07:
SSH keys are not allowed in the environment and only GSSAPI auth is.  So there is a viable reason as key stealing can and has happened ....

As a reference I am using GPG to passphrase protect the key injection from sshpass.

nmelay, 2023-09-18:
> Its a bad practice to put passwords into scripts,

It sure is.

> please use key authentication.

So... what if your SSH server does not support public key authentication?
Many old routers and switches just won't let you do it.

> Won't add it, sorry.

Please think about it again.
This tool does not exist without reason, and its authors are not security illiterates.
Also, sshpass documentation tries to prevent its users from doing stupid things.
Right now, alternatives such as using expect or dropbear are all worse.
sshpass allows you to store your password in a file which, when you think about it, is not that much worse than authenticating with a key stored in a file.

sadtler, 2024-09-08:
Still needed.