Entware/Entware #303: Package request: moreutils

Endareth, 2019-06-16:
Package request: moreutils
---
Surprised this one's missing, seems like it's got so many useful tools in it someone would have prioritized its addition.

- [X] No similar package already added and/or this one got some new killer feature(s),
- [X] Package is in active development (at least, not abandoned for 5 years or more),
- [X] I'm ready to test added package on my {platform} and get a feedback about it.
---
Happy to help test this on a QNAP.

ryzhovau, 2019-06-17:
"[Use more `less` and less `more`](https://en.wikipedia.org/wiki/Less_(Unix))" I heard 20 years ago. 

Why you asked a basic pager, which allows you to scroll (only!) downwards, one page at a time when `less` is available?

Endareth, 2019-06-17:
This is why one should never assume. My apologies, to be more specific I’m referring to the moreutils package as at https://joeyh.name/code/moreutils/
Contains:
chronic: runs a command quietly unless it fails
combine: combine the lines in two files using boolean operations
errno: look up errno names and descriptions
ifdata: get network interface info without parsing ifconfig output
ifne: run a program if the standard input is not empty
isutf8: check if a file or standard input is utf-8
lckdo: execute a program with a lock held
mispipe: pipe two commands, returning the exit status of the first
parallel: run multiple jobs at once
pee: tee standard input to pipes
sponge: soak up standard input and write to a file
ts: timestamp standard input
vidir: edit a directory in your text editor
vipe: insert a text editor into a pipe
zrun: automatically uncompress arguments to command

ryzhovau, 2019-06-17:
Got it!

closed 2019-10-10 by ryzhovau