Entware/Entware #306: Package request:  obfs4

clunietp, 2019-06-21:
Package request
---
https://gitweb.torproject.org/pluggable-transports/obfs4.git/
obfs4proxy to serve obfs4 bridge traffic over the Tor network.

- [x ] No similar package already added and/or this one got some new killer feature(s),
- [x ] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I'm ready to test added package on my armv7 and get a feedback about it.

Platform:  Linux RT-AC68U-2888 2.6.36.4brcmarm #1 SMP PREEMPT Fri May 17 23:40:02 EDT 2019 armv7l ASUSWRT-Merlin

App requires compiling with Go, but I get out of memory errors during compilation on this device.  Thank you

josk0, 2019-07-22:
seconding this request. another use-case: obfs4proxy could also be used to wrap the traffic of an openvpn server. simple-obfs, with the same use-case  is in the repository but the project is deprecated.

clunietp, 2019-08-05:
I was able to get this compiled on my device by the following method:
- Install Go via optware
- git clone from https://gitlab.com/yawning/obfs4.git
- Add a swap file to get around the out of memory error
- remap go packages from '/root/go' to my mounted disk sda1 via `ln -s`  (go was trying to download packages to the internal storage of my device and ran out of disk space)
- build via go

ryzhovau, 2020-02-27:
Done. https://github.com/Entware/entware-go/commit/9e36cfbbbdffa8f5d3294eb72b909e900a050b78

closed 2020-02-27 by ryzhovau