Entware/Entware #309: PTHREAD_ATFORK Missing Symbols

mr-orange, 2019-07-08:
Hi I Have this error using the 2.6.1 version of python-cryptography:

/opt/lib/python3.7/site-packages/cryptography/hazmat/bindings/_openssl.abi3.so: undefined symbol: pthread_atfork

ldd _openssl.abi3.so
        linux-vdso.so.1 (0xbef80000)
        libssl.so.1.0.0 => /opt/lib/libssl.so.1.0.0 (0xb6e3f000)
        libcrypto.so.1.0.0 => /opt/lib/libcrypto.so.1.0.0 (0xb6ccd000)
        libpython3.7.so.1.0 => /opt/lib/libpython3.7.so.1.0 (0xb6a67000)
        libgcc_s.so.1 => /opt/lib/libgcc_s.so.1 (0xb6a4a000)
        libc.so.6 => /opt/lib/libc.so.6 (0xb6908000)
        libdl.so.2 => /opt/lib/libdl.so.2 (0xb68f5000)
        libz.so.1 => /opt/lib/libz.so.1 (0xb68d1000)
        libpthread.so.0 => /opt/lib/libpthread.so.0 (0xb68a8000)
        libutil.so.1 => /opt/lib/libutil.so.1 (0xb6895000)
        libm.so.6 => /opt/lib/libm.so.6 (0xb67d1000)
        /opt/lib/ld-linux.so.3 (0xb6f2c000)

It seem (I'm totally noob here) that the symbol  pthread_atfork is missing from libpthread.so or libgcc_s.so.1

mr-orange, 2019-08-01:
any comments about this?

ryzhovau, 2019-08-21:
Can't reproduce that on `deluge` which uses `python-cryptography`. 

Please, re-install dependent packages, looks like [inconsistent linkage](https://github.com/Entware/Entware/wiki#note-about-packages-upgrade) after last `opkg upgrade`.

closed 2019-10-20 by no-response[bot]