Entware/Entware #320: package: glib2 kernel version question.

lancethepants, 2019-08-14:
Previously for your armv7 repo 2.6.36 kernel variant, glib2 2.26.1 was the latest possible release because newer versions of glib2 required a newer kernel. I thought I remember reading somewhere that this limitation would be removed in a future version glib2.  I now see that your armv7 repo 2.6.36 kernel variant is using the (almost) latest version of glib2. Are newer versions of glib2 able to run on 2.6.36 kernel properly now?

zyxmon, 2019-08-14:
We support 2.6.36 kernel only for arm - nothing changed here. We have armv7 repo with 3.2 kernel and 2.27 glibc.

lancethepants, 2019-08-15:
In one of @ryzhovau previous projects, he mentioned glib2 2.26.1 was the last version it would support because the kernel 2.6.22 was too old to support newer versions of glib. This was a mipsel router albeit.
https://www.linksysinfo.org/index.php?threads/mpd-0-18-5-compilation-error-help.69353/post-238160
Kernel 2.6.36 is a little newer than 2.6.22, but I'm curious if it is newer enough to support version of glib > 2.26.1.

zyxmon, 2019-08-15:
minimal kernel version for glibc 2.26 is 3.2

closed 2019-08-15 by zyxmon

lancethepants, 2019-08-15:
I'm **not** talking about c-library glibc.  I'm talking about glib2, the gnome library.

https://developer.gnome.org/glib/