Entware/Entware #322: opkg Segmentation fault

Mijzelf, 2019-08-18:
The 'new' opkg (2019-05-28) contains a bug. I have a repo, unchanged since 2016, containing kernel modules for a certain family of NASses. The new opkg version segfaults when it tries to read it from /opt/var/opkg-lists/.
After removing the empty fields in the list, everything is OK. Yet I can't reproduce the crash by putting empty fields in /opt/var/opkg-lists/entware, so I'm not sure if it's a combination of problems.

The original Packages.gz triggering the crash is [Packages.gz](https://github.com/Entware/Entware/files/3513024/Packages.gz). The adapted version can be found at http://zyxel.diskstation.eu/Users/Mijzelf/Entware-ng/binaries/armv7/
And a forum entry containing a log is here: http://pogoplug.bplaced.net/zyxelforum/viewtopic.php?f=4&t=86

zyxmon, 2019-08-18:
In case you still use Entware-ng - it is not supported and was not updated for a long time.

Mijzelf, 2019-08-18:
That is not the case. It's Entware armv7-2.6.

zyxmon, 2019-08-18:
You are the first one to report this issue. armv7-2.6 is a popular feed. let us wait for more similar reports with additional info. I am closing the ticket for now.

closed 2019-08-18 by zyxmon

mysteriouss, 2019-10-08:
I'm encountered with the same issue  on x64 Synology DSM6 after running opkg update and hope there'll be a fix, thanks

```
root@DSM6:~# opkg install entware-opt
Installing entware-opt (227000-3) to root...
Downloading https://bin.entware.net/x64-k3.2/entware-opt_227000-3_all.ipk
Configuring entware-opt.
root@DSM6:~# opkg list
Segmentation fault (core dumped)
root@DSM6:~# rm -f /opt/var/opkg-lists/entware
root@DSM6:~# opkg list | wc -l
147

root@DSM6:~# uname -a
Linux DSM6 3.10.102 #15266 SMP Mon Mar 26 15:08:53 CST 2018 x86_64 GNU/Linux synology_broadwell_3617xs
```

zyxmon, 2019-10-09:
It looks like ther is a problem with downloading or unpacking packages list - https://bin.entware.net/x64-k3.2/Packages.gz
Check if /opt/var/opkg-lists/entware is correct.
Synology firmware is used for this operation (wget, gzip). Try with different firmware. You can also try to install busybox (entware version) - but you may need do download https://bin.entware.net/x64-k3.2/Packages (unpacked) to /opt/var/opkg-lists/entware to install busybox (entware version).

mysteriouss, 2019-10-09:
ok, something is wrong with the decompression of Packages.gz file, and the unpacked file Packages  is fine. 

```
root@DSM6:/opt/var/opkg-lists# wget 'https://bin.entware.net/x64-k3.2/Packages' -O entware
--2019-10-09 16:53:12--  https://bin.entware.net/x64-k3.2/Packages
Resolving bin.entware.net... 104.27.176.50, 104.27.177.50, 2606:4700:30::681b:b132, ...
Connecting to bin.entware.net|104.27.176.50|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1236239 (1.2M) [application/octet-stream]
Saving to: 'entware'
100%[=>] 1,236,239   1.28MB/s   in 0.9s
2019-10-09 16:53:15 (1.28 MB/s) - 'entware' saved [1236239/1236239]

root@DSM6:/opt/var/opkg-lists# opkg list | wc -l
149

root@DSM6:/opt/var/opkg-lists# mv entware Package

root@DSM6:/opt/var/opkg-lists# opkg update
Downloading https://bin.entware.net/x64-k3.2/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/entware
root@DSM6:/opt/var/opkg-lists# ls -l
-rw-r--r-- 1 root root  346676 Oct  7 18:46 entware (turns out to be the Packages.gz ?)
-rw-r--r-- 1 root root 1236239 Oct  7 18:46 Package
```

zyxmon, 2019-10-09:
If the problem is with unpacking 
1. Make sure /opt/bin & /opt/sbin are first in your PATH 
2. Install `opkg install gzip`
In this case gzip/gunzip will be used from entware and not from synology

mysteriouss, 2019-10-09:
kind of weird, sorry, it does not work. I've installed gzip manually from the ipk file

```
root@DSM6:~# echo $PATH
/opt/bin:/opt/sbin:/sbin:/bin:/usr/sbin:/usr/bin:/usr/syno/sbin:/usr/syno/bin:/usr/local/sbin:/usr/local/bin

root@DSM6:~# which gzip
/opt/bin/gzip

root@DSM6:~# opkg update
Downloading http://bin.entware.net/x64-k3.2/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/entware
root@DSM6:~# ls -l /opt/var/opkg-lists/
total 340
-rw-r--r-- 1 root root 346676  entware (still Packages.gz)

```

mysteriouss, 2019-10-09:
OK, fixed  by ```opkg install opkg ```...

zyxmon, 2019-10-09:
Thanks. I have updated opkg binary in installation dir (server). New installations will be OK.