Entware/Entware #327: Package request: openfortivpn

mikazlopes, 2019-09-05:
Package request - openfortivpn
---
Allows connecting the routers using Fortigate SSL VPN. Other clients such as OpenSwan and OpenVPN can't offer the same VPN connection type.

- [X] No similar package already added and/or this one got some new killer feature(s),
- [X] Package is in active development (at least, not abandoned for 5 years or more),
- [X] I'm ready to test added package on my {platform} and get feedback about it.

Project available in https://github.com/adrienverge/openfortivpn
---

ryzhovau, 2019-11-03:
Not sure we want to add another VPN client in repo, we'll do it for most popular protocols only.
There's already dozen of VPN solutions in repo, can't add them all, sorry.

Closing for now.

closed 2019-11-03 by ryzhovau

ryzhovau, 2021-02-13:
We've received 3rd request and gave up. `openfortivpn` added recently.