Entware/Entware #329: Package request: Yggdrasil

Lichtenshtein, 2019-09-10:
Package request: Yggdrasil
---
"Yggdrasil is an early-stage implementation of a fully end-to-end encrypted IPv6 network. It is lightweight, self-arranging, supported on multiple platforms and allows pretty much any IPv6-capable application to communicate securely with other Yggdrasil nodes. Yggdrasil does not require you to have IPv6 Internet connectivity - it also works over IPv4.

Although Yggdrasil shares many similarities with cjdns, it employs a different routing algorithm based on a globally-agreed spanning tree and greedy routing in a metric space, and aims to implement some novel local backpressure routing techniques. In theory, Yggdrasil should scale well on networks with internet-like topologies."

- Similar to i2pd, but can work without internet connection at all, as mesh net,

- A branch project from the cjdns developer, but more actively developed

https://github.com/yggdrasil-network/yggdrasil-go
https://yggdrasil-network.github.io/about.html

---

zyxmon, 2019-09-11:
It is always a pain to build application written in go using Entware buildroot.
Try to contact the author to build Yggdrasil yourself for your architecture.

Lichtenshtein, 2019-09-11:
Sry, forgot to mention my platform :\
Platform Xiaomi-Mi Mini/Asus RT-N56U, mipsel

They have .deb packages for mipsel devices
https://yggdrasil-network.github.io/builds.html
I tried to unpack them and run but i guess it won't work this way
```
/opt/home/admin # /opt/etc/yggdrasil -genconf > /opt/etc/yggdrasil.conf
-sh: /opt/etc/yggdrasil: Permission denied
```

zyxmon, 2020-03-02:
I can compile Yggdrasil for testing. Plz write you architecture/feed here/
Package was added - https://github.com/Entware/entware-go/commit/8eeded0681b38941c79939e06334b96ebe1ea0b5

Lichtenshtein, 2020-03-29:
Set for the test. Thank you for adding :)

Platform Asus RT-N56U, mipsel (?)
I'm using mipselsf-k3.4 package list

closed 2020-04-27 by zyxmon