Entware/Entware #336: check_data_file_clashes: Package python-chardet wants to install file /opt/bin/chardetect But that file is already provided by package  * python3-chardet

diman82, 2019-09-21:
Use `--force-overwrite` flag in opkg command. There are only few _dev packages in entware. Please use wiki to install development environment.
Please note we do not support native compilation and recommend cross-compilation.

_Originally posted by @zyxmon in https://github.com/Entware/Entware-ng/issues/559#issuecomment-280229396_

closed 2019-09-21 by diman82

ahmadkarim, 2019-11-19:
@diman82 can you provide some details exactly where to set this flag. Thanks

diman82, 2019-11-19:
@ahmadkarim opkg install _some_package_ **--force-overwrite**

ahmadkarim, 2019-11-19:
thanks but I realize this was a general issue I am having this issue in Yocto build system

diman82, 2019-11-20:
> thanks but I realize this was a general issue I am having this issue in Yocto build system

It shouldn't matter too much, as long it's running Entware