Entware/Entware #342: QNAP install  rtorrent-easy-install fail

ted423, 2019-10-06:
Package request
---
{Describe killer feature of package}

- [ ] No similar package already added and/or this one got some new killer feature(s),
- [ ] Package is in active development (at least, not abandoned for 5 years or more),
- [ ] I'm ready to test added package on my {platform} and get a feedback about it.

---
Please, fill `{}` fields when asking for new package or delete this template in any other case.
```
[/] # opkg install nano  rtorrent-easy-install
Installing nano (4.2-1) to root...
Downloading http://bin.entware.net/x64-k3.2/nano_4.2-1_x64-3.2.ipk
Installing libncursesw (6.1-4) to root...
Downloading http://bin.entware.net/x64-k3.2/libncursesw_6.1-4_x64-3.2.ipk
Installing zlib (1.2.11-3) to root...
Downloading http://bin.entware.net/x64-k3.2/zlib_1.2.11-3_x64-3.2.ipk
Installing libmagic (5.36-1) to root...
Downloading http://bin.entware.net/x64-k3.2/libmagic_5.36-1_x64-3.2.ipk
Installing file (5.36-1) to root...
Downloading http://bin.entware.net/x64-k3.2/file_5.36-1_x64-3.2.ipk
Installing rtorrent-easy-install (0.2-3a) to root...
Downloading http://bin.entware.net/x64-k3.2/rtorrent-easy-install_0.2-3a_x64-3.2.ipk
Installing libopenssl (1.1.1b-5) to root...
Downloading http://bin.entware.net/x64-k3.2/libopenssl_1.1.1b-5_x64-3.2.ipk
Installing ca-bundle (20190110-1) to root...
Downloading http://bin.entware.net/x64-k3.2/ca-bundle_20190110-1_all.ipk
Installing libcurl (7.64.1-1) to root...
Downloading http://bin.entware.net/x64-k3.2/libcurl_7.64.1-1_x64-3.2.ipk
Installing libsigcxx (2.10.1-1) to root...
Downloading http://bin.entware.net/x64-k3.2/libsigcxx_2.10.1-1_x64-3.2.ipk
Installing libtorrent (0.13.7) to root...
Downloading http://bin.entware.net/x64-k3.2/libtorrent_0.13.7_x64-3.2.ipk
Installing xmlrpc-c-common (1.43.08-1) to root...
Downloading http://bin.entware.net/x64-k3.2/xmlrpc-c-common_1.43.08-1_x64-3.2.ipk
Installing xmlrpc-c-internal (1.43.08-1) to root...
Downloading http://bin.entware.net/x64-k3.2/xmlrpc-c-internal_1.43.08-1_x64-3.2.ipk
Installing xmlrpc-c (1.43.08-1) to root...
Downloading http://bin.entware.net/x64-k3.2/xmlrpc-c_1.43.08-1_x64-3.2.ipk
Installing xmlrpc-c-server (1.43.08-1) to root...
Downloading http://bin.entware.net/x64-k3.2/xmlrpc-c-server_1.43.08-1_x64-3.2.ipk
Installing rtorrent-rpc (0.9.7-1) to root...
Downloading http://bin.entware.net/x64-k3.2/rtorrent-rpc_0.9.7-1_x64-3.2.ipk
Installing coreutils (8.30-1) to root...
Downloading http://bin.entware.net/x64-k3.2/coreutils_8.30-1_x64-3.2.ipk
Installing coreutils-stat (8.30-1) to root...
Downloading http://bin.entware.net/x64-k3.2/coreutils-stat_8.30-1_x64-3.2.ipk
Installing coreutils-id (8.30-1) to root...
Downloading http://bin.entware.net/x64-k3.2/coreutils-id_8.30-1_x64-3.2.ipk
Installing gzip (1.10-1) to root...
Downloading http://bin.entware.net/x64-k3.2/gzip_1.10-1_x64-3.2.ipk
Installing curl (7.64.1-1) to root...
Downloading http://bin.entware.net/x64-k3.2/curl_7.64.1-1_x64-3.2.ipk
Installing rutorrent (3.9-1) to root...
Downloading http://bin.entware.net/x64-k3.2/rutorrent_3.9-1_x64-3.2.ipk
Installing rutorrent-plugin-rpc (3.9-1) to root...
Downloading http://bin.entware.net/x64-k3.2/rutorrent-plugin-rpc_3.9-1_x64-3.2.ipk
Installing libncurses (6.1-4) to root...
Downloading http://bin.entware.net/x64-k3.2/libncurses_6.1-4_x64-3.2.ipk
Installing screen (4.6.2-2) to root...
Downloading http://bin.entware.net/x64-k3.2/screen_4.6.2-2_x64-3.2.ipk
Installing libxml2 (2.9.9-1) to root...
Downloading http://bin.entware.net/x64-k3.2/libxml2_2.9.9-1_x64-3.2.ipk
Installing php7 (7.2.17-2) to root...
Downloading http://bin.entware.net/x64-k3.2/php7_7.2.17-2_x64-3.2.ipk
Installing php7-cgi (7.2.17-2) to root...
Downloading http://bin.entware.net/x64-k3.2/php7-cgi_7.2.17-2_x64-3.2.ipk
Installing lighttpd (1.4.53-2) to root...
Downloading http://bin.entware.net/x64-k3.2/lighttpd_1.4.53-2_x64-3.2.ipk
Installing lighttpd-mod-fastcgi (1.4.53-2) to root...
Downloading http://bin.entware.net/x64-k3.2/lighttpd-mod-fastcgi_1.4.53-2_x64-3.2.ipk
Installing lighttpd-mod-scgi (1.4.53-2) to root...
Downloading http://bin.entware.net/x64-k3.2/lighttpd-mod-scgi_1.4.53-2_x64-3.2.ipk
Installing lighttpd-mod-auth (1.4.53-2) to root...
Downloading http://bin.entware.net/x64-k3.2/lighttpd-mod-auth_1.4.53-2_x64-3.2.ipk
Installing lighttpd-mod-authn_file (1.4.53-2) to root...
Downloading http://bin.entware.net/x64-k3.2/lighttpd-mod-authn_file_1.4.53-2_x64-3.2.ipk
Configuring libncursesw.
Configuring zlib.
Configuring libmagic.
Configuring file.
Configuring nano.
Configuring libopenssl.
Configuring ca-bundle.
Configuring libcurl.
Configuring lighttpd.
Configuring lighttpd-mod-auth.
Configuring lighttpd-mod-authn_file.
Configuring xmlrpc-c-common.
Configuring xmlrpc-c-internal.
Configuring xmlrpc-c.
Configuring libxml2.
Configuring php7.
Configuring php7-cgi.
Configuring coreutils.
Configuring coreutils-stat.
Configuring libsigcxx.
Configuring libtorrent.
Configuring xmlrpc-c-server.
Configuring rtorrent-rpc.
Configuring coreutils-id.
Configuring gzip.
Configuring curl.
Configuring rutorrent.
Configuring rutorrent-plugin-rpc.
Configuring libncurses.
Configuring screen.
Configuring lighttpd-mod-fastcgi.
Configuring lighttpd-mod-scgi.
Configuring rtorrent-easy-install.

#### Welcome to rtorrent easy installation script!
 We assuming no lighttpd/rtorrent/rutorrent was installed before!
 Please, answer a few question to configure torrent client.
 Default aswers given in [brackets], just hit Enter to accept it.

Type "username:password" if you want to protect Web UI or hit Enter to leave password protection disabled []: xx:910423P@ssw0rd
Which port should be used by web interface? [81]:
Enter (maximal) download speed in KB/s [2048]:
Enter (maximal) upload speed in KB/s [2048]:
Enter folder name on USB drive where torrent content should be stored [/opt/torrents]:

 Done! You may open 51411-51412 TCP/UDP ports on your router to
 facilitate uploading/donwloading. It's not necessary but recommended.
 rutorrent web UI will be available at following URL:

 http://<ip_address_of_router>:<port>/rutorrent

 Do you wish to start torrent client? [y]:
 Starting lighttpd...              done.
 Starting rtorrent...              failed.
Collected errors:
 * pkg_run_script: package "rtorrent-easy-install" postinst script returned status 255.
 * opkg_configure: rtorrent-easy-install.postinst returned 255.
```

zyxmon, 2019-10-06:
It looks like the configuration that script created is not correct for your device.
Try to run
`/opt/etc/init.d/S85rtorrent start` or even examine S85rtorrent script and start rtorrent manually.
I have installed rtorrent on Qnap 253 NAS without any problems

```
...........
#### Welcome to rtorrent easy installation script!
 We assuming no lighttpd/rtorrent/rutorrent was installed before!
 Please, answer a few question to configure torrent client.
 Default aswers given in [brackets], just hit Enter to accept it.

Type "username:password" if you want to protect Web UI or hit Enter to leave password protection disabled []:
Which port should be used by web interface? [81]:
Enter (maximal) download speed in KB/s [2048]:
Enter (maximal) upload speed in KB/s [2048]:
Enter folder name on USB drive where torrent content should be stored [/opt/torrents]:

 Done! You may open 51411-51412 TCP/UDP ports on your router to
 facilitate uploading/donwloading. It's not necessary but recommended.
 rutorrent web UI will be available at following URL:

 http://<ip_address_of_router>:<port>/rutorrent

 Do you wish to start torrent client? [y]:
 Starting lighttpd...              done.
 Starting rtorrent...              done.

```

closed 2019-10-06 by zyxmon

ted423, 2019-10-06:
> It looks like the configuration that script created is not correct for your device.
> Try to run
> `/opt/etc/init.d/S85rtorrent start` or even examine S85rtorrent script and start rtorrent manually.
> I [...]

I tried /opt/etc/init.d/S85rtorrent start  but it failed,now I use alternative.sh ,it seem rtorrent strat ,but wen show No connection to rTorrent. Check if it is really running. Check $scgi_port and $scgi_host settings in config.php and scgi_port in rTorrent configuration file.

zyxmon, 2019-10-06:
Examine S85rtorrent script, probably screen binary fails to run.
Try to start rtorrent in screen session manually - just look in S85rtorrent, how to do it.
Probably you can ask your questions on qnap forum.
I do not use rtorrent and cannot help you. I see no problems running it on my qnap.

Alternative script installs busybox with more applets, compared with firmware. Probably some is missing on your qnap and standard install.
We do not provide personal support here, sorry.

zyxmon, 2019-10-06:
https://hqt.ro/rutorrent-on-asuswrt-router-through-entware/
This is rather old, but may help.

ted423, 2019-10-06:
> Examine S85rtorrent script, probably screen binary fails to run.
> Try to start rtorrent in screen session manually - just look in S85rtorrent, how to do it.
> Probably you can ask your questions on [...]

Thank's for your help ， it seems the alternative's version ok(but still something wrong). I just think someone maintainence this things here.

yannfill, 2019-11-03:
Starting rtorrent fails because Entware still provides old syntax (pre 0.9.x) rtorrent.conf files. See https://github.com/rakshasa/rtorrent/wiki/rTorrent-0.9-Comprehensive-Command-list-(WIP) to migrate obsolete commands.

xuyue4, 2019-11-23:
> Starting rtorrent fails because Entware still provides old syntax (pre 0.9.x) rtorrent.conf files. See https://github.com/rakshasa/rtorrent/wiki/rTorrent-0.9-Comprehensive-Command-list-(WIP) to migrate obsolete commands.

or might be rtorrent starts(or reports?) very slow. My NAS is aarch64 version and the init script reports failed sometimes , but actually works well with PID. Mystery here :)
 
PS. Screenshot
 
[image]