Entware/Entware #343: Bug: perl-test-harness conflicts with perlbase-app and perlbase-tap

JACK-THINK, 2019-10-09:
platform: `linux-aarch64-k3.10`

Log as below

Collected errors:
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/App/Prove/State/Result/Test.pm
        But that file is already provided by package  * perlbase-app
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/App/Prove/State/Result.pm
        But that file is already provided by package  * perlbase-app
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/App/Prove/State.pm
        But that file is already provided by package  * perlbase-app
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/App/Prove.pm
        But that file is already provided by package  * perlbase-app
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Base.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Formatter/Base.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Formatter/Color.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Formatter/Console/ParallelSession.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Formatter/Console/Session.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Formatter/Console.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Formatter/File/Session.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Formatter/File.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Formatter/Session.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Harness/Env.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Harness.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Object.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Parser/Aggregator.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Parser/Grammar.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Parser/Iterator/Array.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Parser/Iterator/Process.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Parser/Iterator/Stream.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Parser/Iterator.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Parser/IteratorFactory.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Parser/Multiplexer.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Parser/Result/Bailout.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Parser/Result/Comment.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Parser/Result/Plan.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Parser/Result/Pragma.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Parser/Result/Test.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Parser/Result/Unknown.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Parser/Result/Version.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Parser/Result/YAML.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Parser/Result.pm
[... 7 lines cut ...]
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Parser/Scheduler.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Parser/Source.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Parser/SourceHandler/Executable.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Parser/SourceHandler/File.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Parser/SourceHandler/Handle.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Parser/SourceHandler/Perl.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Parser/SourceHandler/RawTAP.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Parser/SourceHandler.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Parser/YAMLish/Reader.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Parser/YAMLish/Writer.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/TAP/Parser.pm
        But that file is already provided by package  * perlbase-tap
 * check_data_file_clashes: Package perl-test-harness wants to install file /opt/lib/perl5/5.28/Test/Harness.pm
        But that file is already provided by package  * perlbase-test
 * opkg_install_cmd: Cannot install package perl-test-harness.

ryzhovau, 2019-10-09:
Thanks for report! This is [5 years old issue](https://github.com/openwrt/packages/commit/2fc7cf44028e370163c16de92aad7a27d1e8f596#diff-a577a7564ab440bb31e775348d99f7de), which came here from upstream. I hope OpenWrt guys will fix it and it will arrive here during next sync. You may override this error as workaround:
```
opkg  --force-overwrite install perl-test-harness
```

ryzhovau, 2019-10-10:
`perl-test-harness` includes content of `perlbase-app`, `perlbase-test`, `perlbase-test` and… nothing more.

I'll close this issue as upstream specific. If OpenWrt guys fix that, a new version arrive here during next sync.

closed 2019-10-10 by ryzhovau