Entware/Entware #345: Tor not starting

pasha-zzz, 2019-10-10:
All was fine for (I think) years. But now (maybe month or 1-2 weeks) tor says INTERNAL ERROR on start and not works.

```
Oct 10 06:35:02.000 [notice] Bootstrapped 0% (starting): Starting
Oct 10 06:35:02.000 [notice] Starting with guard context "default"

============================================================ T= 1570678503
INTERNAL ERROR: Raw assertion failed at src/lib/malloc/map_anon.c:213: nodump_result == 0tor(dump_stack_symbols_to_error_fds+0x34)[0x2a23e940]
tor(tor_raw_assertion_failed_msg_+0x98)[0x2a23ea40]
tor(tor_mmap_anonymous+0x118)[0x2a23f888]
tor(crypto_fast_rng_new_from_seed+0x3c)[0x2a1fa3e8]
tor(crypto_fast_rng_new+0x38)[0x2a1fa4a4]
tor(get_thread_fast_rng+0x58)[0x2a1fa538]
tor(circuit_reset_sendme_randomness+0x34)[0x2a17586c]
tor(+0x207bec)[0x2a207bec]
tor(origin_circuit_new+0x98)[0x2a207d08]
tor(origin_circuit_init+0x2c)[0x2a208954]
tor(circuit_establish_circuit+0x34)[0x2a20e540]
tor(circuit_launch_by_extend_info+0xf0)[0x2a1874c4]
tor(+0x187f34)[0x2a187f34]
tor(connection_ap_handshake_attach_circuit+0x264)[0x2a1887f4]
tor(connection_ap_attach_pending+0x168)[0x2a189104]
Aborted
```

Maybe this error - https://trac.torproject.org/projects/tor/ticket/31570

zyxmon, 2019-10-12:
What feed do you use? (What is your cpu/architecture)

maps2002, 2019-10-12:
I started having the same error, and I am on aarch64-k3.10
Thank you

zyxmon, 2019-10-12:
Plz update tor and check if it works now.

pasha-zzz, 2019-10-13:
All works fine! Thanks. ASUS N18U

closed 2019-10-13 by zyxmon