Entware/Entware #346: package updates and rutorrent

pthall, 2019-10-11:
Packages in Entware were updated October 7, including the dependencies below for rutorrent.

```
opkg - 2019-01-31-d4ba162b-1 - 2019-06-14-dcbc142e-1
libcurl - 7.64.1-1 - 7.66.0-1
lighttpd-mod-authn_file - 1.4.53-2 - 1.4.54-1
xmlrpc-c - 1.43.08-1 - 1.51.03-1
php7-cgi - 7.2.17-2 - 7.2.22-1
coreutils - 8.30-1 - 8.30-2
coreutils-stat - 8.30-1 - 8.30-2
screen - 4.6.2-2 - 4.6.2-3
xmlrpc-c-internal - 1.43.08-1 - 1.51.03-1
libxml2 - 2.9.9-1 - 2.9.9-2
lighttpd - 1.4.53-2 - 1.4.54-1
php7-mod-json - 7.2.17-2 - 7.2.22-1
curl - 7.64.1-1 - 7.66.0-1
coreutils-id - 8.30-1 - 8.30-2
xmlrpc-c-server - 1.43.08-1 - 1.51.03-1
xmlrpc-c-common - 1.43.08-1 - 1.51.03-1
lighttpd-mod-access - 1.4.53-2 - 1.4.54-1
libncurses - 6.1-4 - 6.1-5
lighttpd-mod-accesslog - 1.4.53-2 - 1.4.54-1
php7 - 7.2.17-2 - 7.2.22-1
libncursesw - 6.1-4 - 6.1-5
libyaml - 0.2.1-1 - 0.2.2-1
lighttpd-mod-auth - 1.4.53-2 - 1.4.54-1
gzip - 1.10-1 - 1.10-2
lighttpd-mod-fastcgi - 1.4.53-2 - 1.4.54-1
rtorrent-rpc - 0.9.7-1 - 0.9.8-2
lighttpd-mod-scgi - 1.4.53-2 - 1.4.54-1
libopenssl - 1.1.1b-5 - 1.1.1d-1
```

Prior to the update, rutorrent was working. After the update, rtorrent still works via the CLI. When rutorrent is opened in the browser, there is an error:
>ruTorrent cannot determine the UID of rTorrent user. Some functionality will be unavailable.

[Here is a gif](https://i.imgur.com/879ijpi.gif) of what I think is xmlrpc messages fill the rtorrent CLI. This only appears once rutorrent is opened.

As a test, I started over with a clean entware install and installed only `rtorrent-easy-install`. The same issue still occurred.

Device firmware is DD-WRT v3.0-r40854 on arm7 architecture
```
# uname -a
Linux DD-WRT 4.4.190 #1092 SMP Thu Aug 29 11:32:59 +04 2019 armv7l DD-WRT
```

zyxmon, 2019-10-11:
Probably you can try to downgrade xmlrpc-c package from 1.51.03 to 1.43.08
Old packages are in archive subfolder of repository

pthall, 2019-10-11:
That fixed the xmlrpc filling the CLI.

However, this error persisted
>ruTorrent cannot determine the UID of rTorrent user. Some functionality will be unavailable.

The only way I was able to fix this was to also downgrade `rtorrent-rpc`. I noticed in the downgrade that I needed `libtorrent` and `libsigcxx.` I assume the latest update statically linked these in rtorrent-rpc?

closed 2019-10-14 by ryzhovau