Entware/Entware #349: temperusb package - compilation fails.

escot, 2019-10-17:
`make[3]: Entering directory '/home/arami/Entware/build_dir/target-mipsel_mips32r2_glibc-2.27/temperv14-2.3'
mipsel-openwrt-linux-gnu-gcc -o temperusb temperv14.c -lusb-1.0
cc1: note: someone does not honour COPTS correctly, passed 0 times
temperv14.c:40:10: fatal error: libusb-1.0/libusb.h: No such file or directory
 #include <libusb-1.0/libusb.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Makefile:7: recipe for target 'temperusb' failed
`

ryzhovau, 2019-10-21:
`temperusb` package is not enabled in [Entware configs](https://github.com/Entware/Entware/tree/master/configs). We (or you) should [edit this one](https://github.com/Entware/entware-packages/tree/master/utils/temperusb) first.

ryzhovau, 2019-11-02:
No response, closing.

closed 2019-11-02 by ryzhovau