Entware/Entware #352: "no crontab for root" on Synology DSM-6

blsz, 2019-10-22:
Cron is no longer seeing the file /opt/etc/crontab included with the cron package. 

Entware cron is installed on a Synology DS214 with DSM 6.2.2-24922 Update 3.  I think this problem appeared with DSM 6.  Both entware-ng and entware are affected.   crontab -e creates /opt/var/cron/crontabs/root which is read by crontab -l.  What could have changed in DSM that would prevent  /opt/etc/crontab from being read?  

# crontab -l
no crontab for root

# ls -al /opt/etc/crontab
-rw------- 1 root root 492 Sep 25 07:53 /opt/etc/crontab

# whoami
root

# ps auxf | grep "[c]ron -s"
root      3221  0.0  0.1   4028   560 ?        Ss   11:22   0:00 cron -s

# ls -al /proc/3221/exe
lrwxrwxrwx 1 root root 0 Oct 22 11:22 /proc/3221/exe -> /opt/sbin/cron

# uname -m
armv7l

# opkg list-installed | grep cron
cron - 4.1-3

# cat opkg.conf
src/gz entware http://bin.entware.net/armv7sf-k3.2
dest root /
dest ram /opt/tmp
lists_dir ext /opt/var/opkg-lists
option tmp_dir /opt/tmp
arch all 100
arch armv7-3.x 150
arch armv7-3.2 160

zyxmon, 2019-10-22:
Nothing changed since 2015 - http://forums.zyxmon.org/viewtopic.php?f=5&t=5257 (Russian)
There are two cron versions in Entware
If you have instelled cron package -` /opt/etc/crontab` is used!

closed 2019-10-22 by zyxmon

blsz, 2019-10-23:
I don't understand why crontab -l  (as root) doesn't show the jobs in /opt/etc/crontab    But cron works.  Jobs in /opt/etc/crontab do run.