Entware/Entware #353: All 4.x samba in armv7sf-k3.2 are broken as a matter of fact...

lokapal, 2019-10-23:
Hello, Entware/Optware developers! :)

I've spent a couple of days in attempts of run properly samba at HiMedia Q10 Pro mediaplayer (Android 7.xx). Well, I've found that ALL samba 4.x releases are lame in the tree armv7sf-k3.2:
1. All samba4x-server packages after samba4x-server_4.8.0-1_armv7-3.2.ipk doesn't have nmbd included. It even commented out in /opt/etc/init.d/S91samba. Farewell network discovery and normal samba working!
2. Both samba4x-server_4.8.0-1_armv7-3.2.ipk  and samba4x-server_4.7.3-1_armv7-3.2.ipk doesn't show/copy files that are greater than 2GB. 
3. The bright side: it seems that samba36-server_3.6.25-14_armv7-3.2.ipk is fully functional package (all Windows 10 that have turned OFF SMB2 support will not work with Samba 3.6).

Thanks in advance! It's pure fantastic, I know, but I'd like to know - why arm7hf tree is not compiled by default?

zyxmon, 2019-10-23:
1. https://github.com/Entware/Entware/issues/304
2. This issue was already reported. No solution yet.
3. Samba4 is still may be considered beta.

lokapal, 2019-10-23:
Frankly speaking, I don't understand why it's impossible to compile/create an ADDITIONAL package (like it's done for net tool) for those people who don't use OpenWRT. It easily can be done even by hands (I've tried to use nmbd from 4.0.8.0 with the latest smbd), the single problem is that nmbd has version dependency hardcoded, so adding libraries by hand to /opt/lib cannot help to run the service.

lokapal, 2019-10-23:
https://linoxide.com/linux-how-to/install-specific-version-packages-rhel-centos-7/
Even in the Enterprise environment RedHat/CentOS samba 4.1 is the MINIMAL version allowed to install.

zyxmon, 2019-10-23:
We will add nmbd to samba4 next build since openwrt reverted default option to build nmbd.

The issue is reopened

zyxmon, 2020-02-01:
nmbd added to package

closed 2020-02-01 by zyxmon