Entware/Entware #356: Struggling with make

jonesy2, 2019-11-04:
I'm trying to (re)compile the mpd package, so I can change the encoder options to include lame & ogg, for a ASUS NAS  (x86_64) running Entware.  
I grabbed the docker container for openwrt, followed the Entware Wiki. Apart from some minor things it was going well until I hit 2 errors :-
`$ make package/mpd/compile -j1 V=sc

cp -fpR /home/build/openwrt/Entware/staging_dir/toolchain-x86_64_gcc-7.4.0_glibc-2.27/lib/ld*.so.* /home/build/openwrt/Entware/staging_dir/toolchain-x86_64_gcc-7.4.0_glibc-2.27/lib/ld-unknown.so /home/build/openwrt/Entware/staging_dir/toolchain-x86_64_gcc-7.4.0_glibc-2.27/lib/libmemusage.so /home/build/openwrt/Entware/staging_dir/toolchain-x86_64_gcc-7.4.0_glibc-2.27/lib/libpcprofile.so /home/build/openwrt/Entware/build_dir/target-x86_64_glibc-2.27/toolchain/.pkgdir/libc/opt/lib/
cp: cannot stat '/home/build/openwrt/Entware/staging_dir/toolchain-x86_64_gcc-7.4.0_glibc-2.27/lib/ld*.so.*': No such file or directory
cp: cannot stat '/home/build/openwrt/Entware/staging_dir/toolchain-x86_64_gcc-7.4.0_glibc-2.27/lib/ld-unknown.so': No such file or directory`
The only references google can find relate to cross-compiling, but the referenced files aren't found with apt-get eg glibc.i686
Anyone any ideas how to go forward please?

zyxmon, 2019-11-05:
Entware is not OpenWRT. 
https://github.com/Entware/Entware/wiki/Compile-packages-from-sources

closed 2019-11-05 by zyxmon

jonesy2, 2019-11-05:
I'm trying to compile an Entware package. 
I was hoping you could help with a few pointers as the compilation of the Entware elements are failing.

zyxmon, 2019-11-05:
Just follow the wiki, Do not use openwrt docker - it is useless.

jonesy2, 2019-11-05:
I had the same problem under WSL Ubuntu. I'll try a debian container.
When following the Openwrt instructions,  I follow 1 and 2, then is the next optional step 1 necessary,  and is make menuconfig the correct choice? Or have I misunderstood,  and I just need step 1 before returning to clone Entware.