Entware/Entware #360: how to compile vim with python support?

flowercow, 2019-11-13:
I followed the instructions and set up cross compile environment in debian buster.
I have put --with-feature= huge --enable-python3interp=yes into the Makefile in /feeds/packages/utils/vim. copy vim-fuller section and change to vim-huge
After "make  packages/vim/complie", I got vim-huge_8.xx.ipk and installed into the arm device, but vim --versions  | grep python still output "-python and -python3"

what should I do?

zyxmon, 2019-11-13:
Have you compiled python3 in buildroot environment - `make package/python3/compile` ?
Sometimes extra patches to sources are needed for entware correct compilation.

flowercow, 2019-11-13:
I don't compile Python yet. I will try.

flowercow, 2019-11-13:
> Have you compiled python3 in buildroot environment - `make package/python3/compile` ?
> Sometimes extra patches to sources are needed for entware correct compilation.

I have tried this, and has no effect.

zyxmon, 2019-11-13:
Issues are used for bug reports and package requests.
Personal help can be obtained on forums.
You probably need to modify vim Makefile in Entware and/or make some patches to compile vim with python support. Examples can be found in other packages.

closed 2019-11-13 by zyxmon