Entware/Entware #362: Upgrade to dstat_0.7.4-1 gives error, easily fixable

petterjohanolsen, 2019-11-14:
Hello, using Entware x64 from http://bin.entware.net/x64-k3.2

After upgrade from dstat_0.7.3-1 to dstat_0.7.4-1 it stops working with the following error:

> Traceback (most recent call last):
>   File "/opt/bin/dstat", line 32, in <module>
>     import six

Solution: install python-six with the command
`opkg install python-six`

Python-six should be added as a dependency to the package.

boretom, 2020-02-08:
As far as I can see it's fixed in [entware-oldpackages@5cadd34](https://github.com/Entware/entware-oldpackages-ports/commit/5cadd34ccedbd6a99198ab1c803ed660ab77986f).

ryzhovau, 2020-03-05:
Fixed https://github.com/Entware/entware-oldpackages-ports/commit/5cadd34ccedbd6a99198ab1c803ed660ab77986f

closed 2020-03-05 by ryzhovau