Entware/Entware #364: Package request: expect

MegaCron, 2019-11-18:
Package request
---
- [x] No similar package already added and/or this one got some new killer feature(s),
- [x] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I'm ready to test added package on my {platform} and get a feedback about it.
---
Просмотрел смежное issue #60, где предлагали empty. Он годится лишь для автоматизации. Для последующей интерактивности (передачи управления пользователю после автоматизированного выполнения команд) он не годится, так что хотелось бы увидеть expect в entware и под aarch64 в частности.

Исходники тут https://sourceforge.net/projects/expect/files/Expect/

The-BB, 2020-02-26:
Два варианта:
В [entw](http://bin.entware.net/mipselsf-k3.4/test/entw) - текущая версия tcl из entware. Немного допиленный для сборки expect.
В [owrt](http://bin.entware.net/mipselsf-k3.4/test/owrt) - новая версия tcl из openwrt.
expect c патчами "от соседей" (проект buildroot) + небольшой костыль. Если взлетит, можно будет перебрать патчи от debian||fedora.

The-BB, 2020-02-27:
aarch64:
[tcl](http://bin.entware.net/aarch64-k3.10/test/tcl_8.5.18-1b_aarch64-3.10.ipk)
[expect](http://bin.entware.net/aarch64-k3.10/test/expect_5.45.4-1_aarch64-3.10.ipk)
текущая версия tcl из entware

closed 2020-03-12 by no-response[bot]