Entware/Entware #365: Package request: busybox (full)

MegaCron, 2019-11-19:
Package request
---
- [] No similar package already added and/or this one got some new killer feature(s),
- [x] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I'm ready to test added package on my {platform} and get a feedback about it.
---
Текущий busybox имеет ограниченный набор апплетов, которые при этом ещё и жутко кастрированы. 

Можно собрать более расширенный busybox (в частности, включающий httpd), с апплетами которые поддерживают стандартный, а не укороченный набор ключей ?

ryzhovau, 2019-11-19:
Busybox служит минимальным набором привычных команд для альтернативных установок entware и никак не заменит соответсвующие полнокровные утилиты. Используйте пакеты coreutils-* для того, чтобы заработали все привычные ключи.

closed 2019-11-19 by ryzhovau

zyxmon, 2019-11-19:
httpd можно и добавить. Я некоторое время назад собирал другой микро http сервер для entware, но добавлять не стал в силу жуткой кастрированности.

MegaCron, 2019-11-19:
>Busybox служит минимальным набором привычных команд для альтернативных установок entware

Так я потому и попросил "полный" busybox отдельным пакетом.

> и никак не заменит соответсвующие полнокровные утилиты.

А я не просил "полнокровные" утилиты. Я просил лишь не обрезать функционал тех, что идёт в большинстве сборок busybox. Мне не нужен весь этот ворох coreutils* софта, иной раз с перегруженным функционалом, мне обычно за глаза хватает busybox'овых утилит, но в данном случае они урезаны донельзя. Как пример "полноценного" busybox'a могу привести busybox для ARM* от Стериксона или Мефика. Вот если бы было что-то подобное в entware, мне бы куча пакетов не нужна была :/

zyxmon, 2019-11-19:
coreutils* - это более 100 пакетов, не нужно ставить все, только нужные.
Полноценный ps есть? но не в coreutils, а в procps-ng  - `opkg install procps-ng-ps`
Политика урезания busybox идет от openwrt и мы стараемся ей следовать, т.к. есть пути решения кастрированности апплетов.

zyxmon, 2019-11-19:
Могу для проверки собрать uhttpd - простой сервак из состава openwrt. Для какой архитектуры?

MegaCron, 2019-11-19:
>Полноценный ps есть, но не в coreutils, а в procps-ng - opkg install procps-ng-ps

От того же ps мне нужен именно тот выхлоп, что даёт по дефолту ps из комплекта bb, только без ограничений по символам в выхлопе. Из-за этой мелочи мне нужно ставить лишний пакет, да ещё добавлять в скриптах ключ(и) -ef, дабы был нужный выхлоп. Я же говорю, не нужны мне "полноценные" утилиты. Хотя бы те что есть не кастрируйте по максимуму (и добавьте те, что есть в стандартных/распространённых сборках).

>Могу для проверки собрать uhttpd - простой сервак из состава openwrt. Для какой архитектуры?

Погуглил немного - похоже, он не такой уж и простой - поддержка php7, perl and lua. Можно простенький httpd, для банальных cgi ? Который не требует чтения талмудов-мануалов ?

ryzhovau, 2019-11-19:
На определённые пакеты возложены определённые функции, в частности, busybox в репозитории существуюет для одной цели (см. выше), а coreutils для другой.

Для реализации вашей задачи легковесного http с CGI-поддержкой тоже есть как минимум четыре варианта. Пожалуйста, пользуйтесь.

zyxmon, 2019-11-19:
Я бы даже сказал, что busybox в entware пакет вспомогательный, по умолчанию не ставится (исключение android и кинетик). Установка busybox из entware зачастую "ломает прошивку" - прошивка ожидает свои апплеты, а ей подсовывают другие с другим поведением. Ключи сборки BB идут именно отсюда, из ключей сборки busybox для роутеров (основной сегмент пользователей), чтобы ничего не сломать. При этом все равно, установка busybox-entware на прошивку падавана уже ломает в одном месте поведение.

MegaCron, 2019-11-20:
>На определённые пакеты возложены определённые функции, в частности, busybox в репозитории существует для одной цели (см. выше), а coreutils для другой.

Так я прошу отдельный busybox а не на замену этому. С бОльшим набором апплетов (как у Стериксона например), которые могут быть не "полноценные", но без лишнего  ущемления функционала оных, как сейчас. Для меня busybox - это швейцарский нож, который позволяет решать ворох проблем, без привлечения дополнительных ресурсов. Но похоже разработчикам в данном контексте данный инструмент не интересен. Ну нет, так нет.

>Для реализации вашей задачи легковесного http с CGI-поддержкой тоже есть как минимум четыре варианта. Пожалуйста, пользуйтесь. 
Например ?

>По ним, кстати, куда больше разжеванной документации, чем по апплету httpd из BB.

Так в хелпе апплета вся документация и содержится

>Usage: httpd [-ifv[v]] [-c CONFFILE] [-p [IP:]PORT] [-u USER[:GRP]] [-r REALM] [-h HOME]
>or httpd -d/-e/-m STRING
>
>Listen for incoming HTTP requests
>
>	-i		Inetd mode
>	-f		Don't daemonize
>	-v[v]		Verbose
>	-p [IP:]PORT	Bind to IP:PORT (default *:80)
>	-u USER[:GRP]	Set uid/gid after binding to port
>	-r REALM	Authentication Realm for Basic Authentication
>	-h HOME		Home directory (default .)
>	-c FILE		Configuration file (default {/etc,HOME}/httpd.conf)
>	-m STRING	MD5 crypt STRING
>	-e STRING	HTML encode STRING
>	-d STRING	URL decode STRING

Максимум с чем были проблемы - с прописыванием допусков (в том числе парольных) в конфиге, которые влёт нагуглились на ресурсе посвящённом OpenWrt :D

ryzhovau, 2019-11-20:
Второго Busybox'а не будет, в нём необходимости не видим. Извините.

MegaCron, 2019-11-21:
>Второго Busybox'а не будет, в нём необходимости не видим. Извините.
ОК. Поправить ps по количеству символов в выхлопе и добавить httpd можно ?

zyxmon, 2019-11-21:
httpd - добавим. Если нужны конкретный опции сборки, лучше скажите
```
 CONFIG_BUSYBOX_DEFAULT_FEATURE_HTTPD_RANGES is not set
# CONFIG_BUSYBOX_DEFAULT_FEATURE_HTTPD_SETUID is not set
# CONFIG_BUSYBOX_DEFAULT_FEATURE_HTTPD_BASIC_AUTH is not set
# CONFIG_BUSYBOX_DEFAULT_FEATURE_HTTPD_AUTH_MD5 is not set
# CONFIG_BUSYBOX_DEFAULT_FEATURE_HTTPD_CGI is not set
# CONFIG_BUSYBOX_DEFAULT_FEATURE_HTTPD_CONFIG_WITH_SCRIPT_INTERPR is not set
# CONFIG_BUSYBOX_DEFAULT_FEATURE_HTTPD_SET_REMOTE_PORT_TO_ENV is not set
# CONFIG_BUSYBOX_DEFAULT_FEATURE_HTTPD_ENCODE_URL_STR is not set
# CONFIG_BUSYBOX_DEFAULT_FEATURE_HTTPD_ERROR_PAGES is not set
# CONFIG_BUSYBOX_DEFAULT_FEATURE_HTTPD_PROXY is not set
# CONFIG_BUSYBOX_DEFAULT_FEATURE_HTTPD_GZIP is not set
```
ps менять опасно, выше написал почему. Если готовы протестировать - говорите для какой архитектуры собрать - выложу для теста (httpd).

MegaCron, 2019-11-23:
Под платформу mips.

Нужны опции :

FEATURE_HTTPD_RANGES
FEATURE_HTTPD_SETUID
FEATURE_HTTPD_BASIC_AUTH
FEATURE_HTTPD_AUTH_MD5
FEATURE_HTTPD_CGI
FEATURE_HTTPD_ENCODE_URL_STR
FEATURE_HTTPD_ERROR_PAGES

Опция FEATURE_HTTPD_CONFIG_WITH_SCRIPT_INTERPR в целом бы пригодилась, но не критично.
На опцию FEATURE_HTTPD_PROXY интересно было бы поглядеть т.к. помню, сколько промаялся с реверс-прокси на lighttpd, но так же не критично.
Опция FEATURE_HTTPD_GZIP определённо не нужна
Где применять переменную REMOTE_PORT, что идёт в env-выхлопе из под cgi и за которую отвечает опция FEATURE_HTTPD_SET_REMOTE_PORT_TO_ENV я так и не понял, так что она в целом тоже не нужна. 

Как-то так :)

zyxmon, 2019-11-23:
>> Под платформу mips.

Уточню на всякий - mips (мало у кого) или mipsel (некоторые путают). Просто тулчейн для mipsel, всех arm и intel собран -для этих соберу без проблем. Для mips придется ждать следующего обновления.

MegaCron, 2019-11-23:
>Уточню на всякий - mips (мало у кого) или mipsel (некоторые путают). Просто тулчейн для mipsel, всех arm и intel собран -для этих соберу без проблем. Для mips придется ждать следующего обновления.
В выхлопе uname значится mips GNU/Linux, в cpuinfo :
>cpu model               : MIPS 24Kc V7.12  FPU V0.0

zyxmon, 2019-11-23:
В выхлопе и у mips, и у mipsel будет одно и то же. Поэтому многие и путаются. Entware то стоит? На mipsel не поставить вариант для mips и наоборот!

MegaCron, 2019-11-23:
>В выхлопе и у mips, и у mipsel будет одно и то же. Поэтому многие и путаются.

Похоже я попал именно в эту категорию людей :)

>Entware то стоит?

Стоит. 

В  /opt/etc/opkg.conf значится ссылка на mipsel-репозиторий. Не  mips. Сейчас смутно припоминаю, что скрипт который я взял при установке entware поначалу, не "пошёл", далее взял скрипт под mipsel и всё "завелось"

zyxmon, 2019-11-24:
Проверяйте - http://test.entware.net/mipselsf-k3.4/zyxmon/busybox_1.31.0-1a_mipsel-3.4.ipk
Добавил httpd и длинную опцию ps. Если работает - ставьте на hold, если нет - пишите, в чем проблема.

MegaCron, 2019-11-28:
Извиняюсь за задержку.

Похоже малой кровью обойтись не удастся и надо вкорячивать вместе  со всеми зависимостями ?

>-bash-4.4# busybox-1.31
>-bash: /tmp/tools/bin/busybox-1.31: No such file or directory
>-bash-4.4# strace busybox-1.31
>execve("/tmp/bin/busybox-1.31", ["busybox-1.31"], [/* 28 vars */]) = -1 ENOENT (No such file or directory)
>dup(2)                                  = 3
>fcntl64(3, F_GETFL)                     = 0x2 (flags O_RDWR)
>fstat64(3, {st_mode=S_IFCHR|0620, st_rdev=makedev(136, 0), ...}) = 0
>old_mmap(NULL, 65536, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x2ac3a000
>_llseek(3, 0, 0x7ffdee68, SEEK_CUR)     = -1 ESPIPE (Illegal seek)
>write(3, "strace: exec: No such file or dir"..., 40strace: exec: No such file or directory
>) = 40
>close(3)                                = 0
>munmap(0x2ac3a000, 65536)               = 0
>exit_group(1)                           = ?
>-bash-4.4#

Хотел проверить бинарник с минимальным вмешательством, но похоже не выйдет и придётся его ставить "классическим" способом (вот тут я боялся, что после этого что-то поломается). Или его надо было закинуть в /opt/bin ? Хм. В любом случае сначала сделаю бэкап, а после продолжу тесты.