Entware/Entware #367: Package request: cloud-print-connector

sxiii, 2019-11-21:
Package request
---
Cloud-print-connector allows you to use Android phone or your google account to send documents to a connected printer.

- [+] No similar package already added and/or this one got some new killer feature(s),
- [+] Package is in active development (at least, not abandoned for 5 years or more),
- [+] I'm ready to test added package on my {platform} and get a feedback about it.

---

ryzhovau, 2019-11-21:
Did you tried `python-cloudprint`?

sxiii, 2019-11-22:
@ryzhovau о, спасибо! А есть информация по тому что будет работать на OpenWRT MIPS32EL и как его туда "добыть" лучше всего? Ещё раз спасибо!

ryzhovau, 2019-11-22:
We don't provide personal support here, sorry.

`python-cloudprint` is working on MIPS ([checked](https://pastebin.com/VT3eV5xi) 7 years ago), all print jobs will arrive as PDF files. You need to set up CUPS for PDF printing.

closed 2019-11-22 by ryzhovau