Entware/Entware #368: Package request: zbarimg

MegaCron, 2019-11-21:
Package request
---
- [x] No similar package already added and/or this one got some new killer feature(s),
- [] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I'm ready to test added package on my {platform} and get a feedback about it.

---
 Утилита для декодирования QR-кодов. Для энкода (то бишь создания) QR-кодов в entware софт есть, а для декода - нет.

http://zbar.sourceforge.net
http://sourceforge.net/projects/zbar/files/zbar/0.10/zbar-0.10.tar.bz2/download

zyxmon, 2019-11-21:
Я смотрю у клона репы с sourceforge на github более 800 форков. Есть современное развитие проекта?

MegaCron, 2019-11-22:
>Я смотрю у клона репы с sourceforge на github более 800 форков. Есть современное развитие проекта?

К сожалению, не в курсе :(

pasha-zzz, 2019-11-22:
> Я смотрю у клона репы с sourceforge на github более 800 форков. Есть современное развитие проекта?

https://github.com/mchehab/zbar

closed 2020-01-30 by The-BB