Entware/Entware #370: [alt, armv7sf-k3.2] OPKG does not detect package lists

ValdikSS, 2019-11-24:
OPKG which is downloaded using alternative installation script for armv7sf-k3.2 does not detect any packages in the package list. The installation fails on `opkg install busybox` step. Execution of `/opt/bin/opkg --lists-dir /opt/var/opkg-lists/ -A list` does not produce any output.

```
/opt/bin/opkg --lists-dir /opt/var/opkg-lists/ --verbosity=4 -A list
opkg_conf_parse_file: Loading conf file /opt/etc/opkg.conf.
opkg_conf_parse_file: Supported arch all priority (100)
opkg_conf_parse_file: Supported arch armv7-3.x priority (150)
opkg_conf_parse_file: Supported arch armv7-3.2 priority (160)
pkg_hash_load_feeds: 
pkg_hash_load_status_files:
hash_table: pkg-hash, 12288 bytes
 n_buckets=1024, n_elements=0, n_collisions=0
 max_bucket_len=0, n_used_buckets=0, ave_bucket_len=0.00
 n_hits=0, n_misses=0
hash_table: file-hash, 12288 bytes
 n_buckets=1024, n_elements=0, n_collisions=0
 max_bucket_len=0, n_used_buckets=0, ave_bucket_len=0.00
 n_hits=0, n_misses=0
hash_table: obs-file-hash, 768 bytes
 n_buckets=64, n_elements=0, n_collisions=0
 max_bucket_len=0, n_used_buckets=0, ave_bucket_len=0.00
 n_hits=0, n_misses=0
```

The `pkg_hash_load_feeds` and `pkg_hash_load_status_files` are blank for some reason. The list itself, `/opt/var/opkg-lists/entware`, doesn't seem to be corrupt and could be gunzip'ed successfully, with no change in opkg behavior.

This is Huawei E5770 LTE router, with Android kernel 3.4.5 and hacked up Android userspace. Entware used to work fine on it half a year ago, without any issues.
Full output of alternative installation script with `set -x`:

```
/opt # source ./alternative.sh 
+ source ./alternative.sh
+ unset LD_LIBRARY_PATH
+ unset LD_PRELOAD
+ echo 'Info: Checking for prerequisites and creating folders...'
Info: Checking for prerequisites and creating folders...
+ '[' -d /opt ]
+ echo 'Warning: Folder /opt exists!'
Warning: Folder /opt exists!
+ '[' -d /opt/bin ]
+ mkdir -p /opt/bin
+ '[' -d /opt/etc ]
+ mkdir -p /opt/etc
+ '[' -d /opt/lib/opkg ]
+ mkdir -p /opt/lib/opkg
+ '[' -d /opt/tmp ]
+ mkdir -p /opt/tmp
+ '[' -d /opt/var/lock ]
+ mkdir -p /opt/var/lock
+ echo 'Info: Opkg package manager deployment...'
Info: Opkg package manager deployment...
+ DLOADER=ld-linux.so.3
+ URL=http://bin.entware.net/armv7sf-k3.2/installer
+ wget http://bin.entware.net/armv7sf-k3.2/installer/opkg -O /opt/bin/opkg
Connecting to bin.entware.net (104.27.176.50:80)
opkg                 100% |**************************************************************************************************|  135k  0:00:00 ETA
+ chmod 755 /opt/bin/opkg
+ wget http://bin.entware.net/armv7sf-k3.2/installer/opkg.conf -O /opt/etc/opkg.conf
Connecting to bin.entware.net (104.27.176.50:80)
opkg.conf            100% |**************************************************************************************************|   190  0:00:00 ETA
+ wget http://bin.entware.net/armv7sf-k3.2/installer/ld-2.27.so -O /opt/lib/ld-2.27.so
Connecting to bin.entware.net (104.27.176.50:80)
ld-2.27.so           100% |**************************************************************************************************|  135k  0:00:00 ETA
+ wget http://bin.entware.net/armv7sf-k3.2/installer/libc-2.27.so -O /opt/lib/libc-2.27.so
Connecting to bin.entware.net (104.27.176.50:80)
libc-2.27.so         100% |**************************************************************************************************| 1218k  0:00:00 ETA
+ wget http://bin.entware.net/armv7sf-k3.2/installer/libgcc_s.so.1 -O /opt/lib/libgcc_s.so.1
Connecting to bin.entware.net (104.27.176.50:80)
libgcc_s.so.1        100% |**************************************************************************************************| 50840  0:00:00 ETA
+ wget http://bin.entware.net/armv7sf-k3.2/installer/libpthread-2.27.so -O /opt/lib/libpthread-2.27.so
Connecting to bin.entware.net (104.27.176.50:80)
libpthread-2.27.so   100% |**************************************************************************************************| 92648  0:00:00 ETA
+ cd /opt/lib
+ chmod 755 ld-2.27.so
+ ln -s ld-2.27.so ld-linux.so.3
+ ln -s libc-2.27.so libc.so.6
+ ln -s libpthread-2.27.so libpthread.so.0
+ echo 'Info: Basic packages installation...'
Info: Basic packages installation...
+ /opt/bin/opkg update
Downloading http://bin.entware.net/armv7sf-k3.2/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/entware
+ /opt/bin/opkg install busybox
Unknown package 'busybox'.
Collected errors:
 * opkg_install_cmd: Cannot install package busybox.
+ /opt/bin/opkg install entware-opt
Unknown package 'entware-opt'.
Collected errors:
 * opkg_install_cmd: Cannot install package entware-opt.
+ chmod 777 /opt/tmp
+ cp -f /opt/etc/passwd.1 /opt/etc/passwd
cp: can't stat '/opt/etc/passwd.1': No such file or directory
+ cp -f /opt/etc/group.1 /opt/etc/group
cp: can't stat '/opt/etc/group.1': No such file or directory
+ cp -f /opt/etc/shells.1 /opt/etc/shells
cp: can't stat '/opt/etc/shells.1': No such file or directory
+ '[' -f /etc/localtime ]
+ echo 'Info: Congratulations!'
Info: Congratulations!
+ echo 'Info: If there are no errors above then Entware was successfully initialized.'
Info: If there are no errors above then Entware was successfully initialized.
+ echo 'Info: Add /opt/bin & /opt/sbin to your PATH variable'
Info: Add /opt/bin & /opt/sbin to your PATH variable
+ echo 'Info: Add '"'"'/opt/etc/init.d/rc.unslung start'"'"' to firmware startup script for Entware services to start'
Info: Add '/opt/etc/init.d/rc.unslung start' to firmware startup script for Entware services to start
+ echo 

+ echo 'This is an alternative Entware installation. We recomend to install and setup Entware version of ssh server'
This is an alternative Entware installation. We recomend to install and setup Entware version of ssh server
+ echo 'and use it instead of a firmware supplied one. You can install dropbear or openssh as an ssh server'
and use it instead of a firmware supplied one. You can install dropbear or openssh as an ssh server
```

zyxmon, 2019-11-24:
Трудно сказать в чем дело. Начнем с предположения, что или Packages.gz криво скачался или (такое впервые, но очень похоже) утилиты Вашего роутера не могут распаковать этот архив или просто отсутствуют.
Должно быть так (если все скачалось)
```
 ls -al /opt/var/o
pkg-lists/
drwxr-xr-x    2 root     root          4096 Nov 24 22:55 .
drwxr-xr-x    8 root     root          4096 Sep 25 17:53 ..
-rw-rw-rw-    1 root     root        288373 Nov 24 22:55 entware

```

ValdikSS, 2019-11-24:
Все скачалось корректно, `zcat /opt/var/opkg-lists/entware` выдает список пакетов. Я пытался его подсунуть в распакованном виде — результат аналогичный. Я, на всякий случай, удостоверился, что используется `ld` от entware, и что LD_LIBRARY_PATH правильный. Сами команды opkg визуально работают корректно, но файлы пакетов не могут обнаружить.

Strace запустить не могу.

zyxmon, 2019-11-24:
`/opt/bin/opkg --lists-dir /opt/var/opkg-lists/ --verbosity=4 -A list` выдает
```
opkg_conf_parse_file: Loading conf file /opt/etc/opkg.conf.
opkg_conf_parse_file: Supported arch all priority (100)
opkg_conf_parse_file: Supported arch armv7-3.x priority (150)
opkg_conf_parse_file: Supported arch armv7-3.2 priority (160)
pkg_hash_load_feeds:
flag_related_packages: propagating pkg flag to dependent abpkg libc
flag_related_packages: propagating pkg flag to dependent abpkg libssp
flag_related_packages: propagating pkg flag to dependent abpkg librt
flag_related_packages: propagating pkg flag to dependent abpkg libpthread
flag_related_packages: propagating pkg flag to dependent abpkg libusb-compat
pkg_vec_insert_merge: Adding new pkg=25volt version=1.0.5 arch=armv7-3.2.
pkg_vec_insert_merge: Adding new pkg=3proxy version=0.8.13 arch=armv7-3.2.
pkg_vec_insert_merge: Adding new pkg=6relayd version=2013-12-10.1 arch=armv7-3.2.
flag_related_packages: propagating pkg flag to dependent abpkg libopenssl
flag_related_packages: propagating pkg flag to dependent abpkg libpcre

```
Пока в голове мыслей нет, что не так. LD_LIBRARY_PATH и LD_PRELOAD не должны быть заданы.

zyxmon, 2019-11-24:
А может попробовать скачать busybox и установить локально, /opt/bin в начало PATH и проверить, что opkg заработал?
```
wget http://bin.entware.net/armv7sf-k3.2/busybox_1.31.0-1_armv7-3.2.ipk
/opt/bin/opkg install ./busybox_1.31.0-1_armv7-3.2.ipk
```

ValdikSS, 2019-11-24:
```
wget http://bin.entware.net/armv7sf-k3.2/busybox_1.31.0-1_armv7-3.2.ipk
/opt/bin/opkg install ./busybox_1.31.0-1_armv7-3.2.ipk
```

```
# /opt/bin/opkg --verbosity=4 install ./busybox_1.31.0-1_armv7-3.2.ipk

opkg_conf_parse_file: Loading conf file /opt/etc/opkg.conf.
opkg_conf_parse_file: Supported arch all priority (100)
opkg_conf_parse_file: Supported arch armv7soft priority (150)
opkg_conf_parse_file: Supported arch armv7-2.6 priority (160)
opkg_install_cmd: ./busybox_1.31.0-1_armv7-3.2.ipk
pkg_init_from_file: applying abpkg flag to ./busybox_1.31.0-1_armv7-3.2.ipk
Segmentation fault
```

```
# gdb ./opkg ./core

GNU gdb (GDB) Fedora 8.3-6.fc30
…
(gdb) set solib-search-path lib
Reading symbols from /home/valdikss/opkg/lib/libpthread-2.23.so...
(No debugging symbols found in /home/valdikss/opkg/lib/libpthread-2.23.so)
warning: Unable to find libthread_db matching inferior's thread library, thread debugging will not be available.
Reading symbols from /home/valdikss/opkg/lib/libgcc_s.so.1...
(No debugging symbols found in /home/valdikss/opkg/lib/libgcc_s.so.1)
Reading symbols from /home/valdikss/opkg/lib/libc-2.23.so...
(No debugging symbols found in /home/valdikss/opkg/lib/libc-2.23.so)
Reading symbols from /home/valdikss/opkg/lib/ld-2.23.so...
(No debugging symbols found in /home/valdikss/opkg/lib/ld-2.23.so)
(gdb) bt
#0  0xb6e83c18 in ?? () from /home/valdikss/opkg/lib/libc-2.23.so
#1  0xb6e6d804 in ?? () from /home/valdikss/opkg/lib/libc-2.23.so
#2  0xb6e05fac in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

zyxmon, 2019-11-24:
А если заменить бинарник opkg на старый - http://bin.entware.net/armv7sf-k3.2/installer/opkg.old

ValdikSS, 2019-11-24:
Стоп, прошу прощения, падение вызвано моей попыткой установки версии k2.6, с целью проверки проблемы (всё то же самое). И, вообще говоря, это странно — падать не должно.

С k3.2:
```
# /opt/bin/opkg install --verbosity=4 busybox_1.31.0-1_armv7-3.2.ipk 
opkg_conf_parse_file: Loading conf file /opt/etc/opkg.conf.
opkg_conf_parse_file: Supported arch all priority (100)
opkg_conf_parse_file: Supported arch armv7-3.x priority (150)
opkg_conf_parse_file: Supported arch armv7-3.2 priority (160)
opkg_install_cmd: busybox_1.31.0-1_armv7-3.2.ipk
pkg_init_from_file: applying abpkg flag to busybox_1.31.0-1_armv7-3.2.ipk
hash_table: pkg-hash, 12288 bytes
        n_buckets=1024, n_elements=0, n_collisions=0
        max_bucket_len=0, n_used_buckets=0, ave_bucket_len=0.00
        n_hits=0, n_misses=0
hash_table: file-hash, 12288 bytes
        n_buckets=1024, n_elements=0, n_collisions=0
        max_bucket_len=0, n_used_buckets=0, ave_bucket_len=0.00
        n_hits=0, n_misses=0
hash_table: obs-file-hash, 768 bytes
        n_buckets=64, n_elements=0, n_collisions=0
        max_bucket_len=0, n_used_buckets=0, ave_bucket_len=0.00
        n_hits=0, n_misses=0
Collected errors:
 * pkg_init_from_file: Malformed package file busybox_1.31.0-1_armv7-3.2.ipk.
```

ValdikSS, 2019-11-24:
> А если заменить бинарник opkg на старый - http://bin.entware.net/armv7sf-k3.2/installer/opkg.old

Старая версия работает.

```
/opt # /opt/bin/opkg install --verbosity=4 busybox_1.31.0-1_armv7-3.2.ipk 
opkg_conf_parse_file: Loading conf file /opt/etc/opkg.conf.
opkg_conf_parse_file: Supported arch all priority (100)
opkg_conf_parse_file: Supported arch armv7-3.x priority (150)
opkg_conf_parse_file: Supported arch armv7-3.2 priority (160)
opkg_install_cmd: busybox_1.31.0-1_armv7-3.2.ipk
pkg_init_from_file: applying abpkg flag to busybox_1.31.0-1_armv7-3.2.ipk
hash_table: pkg-hash, 12288 bytes
        n_buckets=1024, n_elements=0, n_collisions=0
        max_bucket_len=0, n_used_buckets=0, ave_bucket_len=0.00
        n_hits=0, n_misses=0
hash_table: file-hash, 12288 bytes
        n_buckets=1024, n_elements=0, n_collisions=0
        max_bucket_len=0, n_used_buckets=0, ave_bucket_len=0.00
        n_hits=0, n_misses=0
hash_table: obs-file-hash, 768 bytes
        n_buckets=64, n_elements=0, n_collisions=0
        max_bucket_len=0, n_used_buckets=0, ave_bucket_len=0.00
        n_hits=0, n_misses=0
Collected errors:
 * pkg_init_from_file: Malformed package file busybox_1.31.0-1_armv7-3.2.ipk.
/opt # ^C
/opt # ls
bin                             etc                             tmp                             var
busybox_1.31.0-1_armv7-3.2.ipk  lib                             usr
/opt # cd bin/
/opt/bin # ls
opkg
/opt/bin # wget http://bin.entware.net/armv7sf-k3.2/installer/opkg.old
Connecting to bin.entware.net (104.27.176.50:80)
opkg.old             100% |***************************************************************************************************************************************************************|  131k  0:00:00 ETA
/opt/bin # chmod +x opkg.old 
/opt/bin # cd ..
/opt # ls
bin                             etc                             tmp                             var
busybox_1.31.0-1_armv7-3.2.ipk  lib                             usr
/opt # /opt/bin/opkg.old install ./busybox_1.31.0-1_armv7-3.2.ipk 
Installing busybox (1.31.0-1) to root...
Installing libc (2.27-8) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/libc_2.27-8_armv7-3.2.ipk
Installing libgcc (7.4.0-8) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/libgcc_7.4.0-8_armv7-3.2.ipk
Installing libssp (7.4.0-8) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/libssp_7.4.0-8_armv7-3.2.ipk
Installing librt (2.27-8) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/librt_2.27-8_armv7-3.2.ipk
Installing libpthread (2.27-8) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/libpthread_2.27-8_armv7-3.2.ipk
Configuring libgcc.
Configuring libc.
Configuring libpthread.
Configuring libssp.
Configuring librt.
Configuring busybox.
```

ValdikSS, 2019-11-24:
Но `/opt/bin/opkg.old list` падает:

```
(gdb) bt
#0  0xb6ec8e24 in strlen () from /home/valdikss/opkg/lib/libc-2.27.so
#1  0x0001f27c in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

zyxmon, 2019-11-24:
А если взять "свежие либы" (тулчейн то поменялся немного) отсюда
http://bin.entware.net/armv7sf-k3.2/installer/new/

zyxmon, 2019-11-24:
Похоже обновление инсталляционного бинарника opkg без библиотек к этому и привело.
Просьба проверить установку с либами из http://bin.entware.net/armv7sf-k3.2/installer/new/ и "новым" http://bin.entware.net/armv7sf-k3.2/installer/opkg (с тем, что был изначально)

ValdikSS, 2019-11-24:
> А если взять "свежие либы" (тулчейн то поменялся немного) отсюда
> http://bin.entware.net/armv7sf-k3.2/installer/new/

Результат аналогичный, что для opkg, что для opkg.old.

```
(gdb) bt
#0  0xb6e45e24 in strlen () from /home/valdikss/opkg/lib/libc-2.27.so
#1  0x0001f27c in ?? ()

(gdb) x/8x $sp
0xbe875b20:     0xffbfffff      0x00000000      0xaaaaaaab      0x00000002
0xbe875b30:     0x0000020a      0x00001059      0xb6f10250      0x00000007

(gdb) x $r0
0xfffffff8:     Cannot access memory at address 0xfffffff8
(gdb) x $r1
0x0:    Cannot access memory at address 0x0
(gdb) x $r2
0x80000000:     Cannot access memory at address 0x80000000
(gdb) x $r3
0x0:    Cannot access memory at address 0x0

(gdb) x/10i $pc
=> 0xb6e45e24 <strlen+36>:      ldrd    r2, r3, [r1]
   0xb6e45e28 <strlen+40>:      pld     [r1, #64]       ; 0x40
   0xb6e45e2c <strlen+44>:      add.w   r0, r0, #8
   0xb6e45e30 <strlen+48>:      uadd8   r2, r2, r12
   0xb6e45e34 <strlen+52>:      sel     r2, r4, r12
   0xb6e45e38 <strlen+56>:      uadd8   r3, r3, r12
   0xb6e45e3c <strlen+60>:      sel     r3, r2, r12
   0xb6e45e40 <strlen+64>:      cbnz    r3, 0xb6e45e96 <strlen+150>
```

ValdikSS, 2019-11-24:
На всякий случай, удалилил /opt, скачал альтернативный установщик, заменил ссылку (URL) на /new, поправил пути у opkg, переустановил — всё то же самое, opkg list не выводит список пакетов.

zyxmon, 2019-11-24:
Будем думать. Пытаться повторить. У меня есть две armv7 железки.
А пока ставьте с opkg.old, может быть стоит сделать hold libc libgcc libpthread opkg (пока все работает). Хотя в процессе установки все обновляется, и, возможно, перестает работать.

ValdikSS, 2019-11-24:
`opkg.old` также не работает полностью корректно — падение при вызове `opkg list` (хоть и пакет через `opkg install` установить возможно)

ValdikSS, 2019-11-24:
Протестировал тот же самый установщик на одноплатнике с ядром 3.4.113 и Ubuntu 16.04 — устанавливается и работает корректно. Буду пробовать отлаживать на Huawei дальше.

ValdikSS, 2019-11-25:
opkg пытается распаковать лист с помощью вызова системного `gzip`, и не выдает ошибку, если gzip отсутствует в системе.

```
[pid  4424] execve("/bin/gzip", ["gzip", "-d", "-c", "/opt/var/opkg-lists/entware"], 0xbecf8e50 /* 9 vars */) = -1 ENOENT (No such file or directory)
[pid  4424] execve("/sbin/gzip", ["gzip", "-d", "-c", "/opt/var/opkg-lists/entware"], 0xbecf8e50 /* 9 vars */) = -1 ENOENT (No such file or directory)
[pid  4424] execve("/app/bin/gzip", ["gzip", "-d", "-c", "/opt/var/opkg-lists/entware"], 0xbecf8e50 /* 9 vars */) = -1 ENOENT (No such file or directory)
[pid  4424] execve("/system/sbin/gzip", ["gzip", "-d", "-c", "/opt/var/opkg-lists/entware"], 0xbecf8e50 /* 9 vars */) = -1 ENOENT (No such file or directory)
[pid  4424] execve("/system/bin/gzip", ["gzip", "-d", "-c", "/opt/var/opkg-lists/entware"], 0xbecf8e50 /* 9 vars */) = -1 ENOENT (No such file or directory)
[pid  4424] execve("/system/xbin/gzip", ["gzip", "-d", "-c", "/opt/var/opkg-lists/entware"], 0xbecf8e50 /* 9 vars */) = -1 ENOENT (No such file or directory)
[pid  4424] exit_group(-1)              = ?
```

Решил проблему враппер-скриптом `/bin/gzip`, запускающим gunzip из busybox. Теперь всё работает корректно с альтернативным инсталлятором с сайта, без его модификации или библиотек из /new.

closed 2019-11-25 by ValdikSS