Entware/Entware #371: request package python3-readline

listenfree, 2019-11-26:
Package request
---
{aarch64-k3.10,need python3-readline,the new python3.7x without readline package,when interact with python3,press up down left right not working,return ^[[A^[[C^[[D}

- [ ] No similar package already added and/or this one got some new killer feature(s),
- [ ] Package is in active development (at least, not abandoned for 5 years or more),
- [ ] I'm ready to test added package on my {platform} and get a feedback about it.

---
Please, fill `{}` fields when asking for new package or delete this template in any other case.

zyxmon, 2019-11-26:
You can add it yourself - https://github.com/Entware/Entware/wiki/Self-installation-of-python-modules

listenfree, 2019-11-26:
running on my arm synology nas ds119j,please help

admin@fisher:/opt/bin$ python3 pip install readline
Collecting readline
  Using cached https://files.pythonhosted.org/packages/f4/01/2cf081af8d880b44939a5f1b446551a7f8d59eae414277fd0c303757ff1b/readline-6.2.4.1.tar.gz
Installing collected packages: readline
    Running setup.py install for readline ... error
    ERROR: Command errored out with exit status 1:
     command: /opt/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/opt/tmp/pip-install-iwwqg0oj/readline/setup.py'"'"'; __file__='"'"'/opt/tmp/pip-install-iwwqg0oj/readline/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /opt/tmp/pip-record-2ccdnu7n/install-record.txt --single-version-externally-managed --compile
         cwd: /opt/tmp/pip-install-iwwqg0oj/readline/
    Complete output (230 lines):
    
    ============ Building the readline library ============
    
    readline-6.2/
    readline-6.2/doc/
    readline-6.2/doc/Makefile.in
    readline-6.2/doc/texinfo.tex
    readline-6.2/doc/version.texi
    readline-6.2/doc/fdl.texi
    readline-6.2/doc/rlman.texi
    readline-6.2/doc/rltech.texi
    readline-6.2/doc/rluser.texi
    readline-6.2/doc/rluserman.texi
    readline-6.2/doc/history.texi
    readline-6.2/doc/hstech.texi
    readline-6.2/doc/hsuser.texi
    readline-6.2/doc/readline.3
    readline-6.2/doc/history.3
    readline-6.2/doc/texi2dvi
    readline-6.2/doc/texi2html
    readline-6.2/doc/readline.ps
    readline-6.2/doc/history.ps
    readline-6.2/doc/rluserman.ps
    readline-6.2/doc/readline.dvi
    readline-6.2/doc/history.dvi
    readline-6.2/doc/rluserman.dvi
    readline-6.2/doc/readline.info
    readline-6.2/doc/history.info
    readline-6.2/doc/rluserman.info
    readline-6.2/doc/readline.html
    readline-6.2/doc/history.html
    readline-6.2/doc/rluserman.html
    readline-6.2/doc/readline.0
    readline-6.2/doc/history.0
    readline-6.2/doc/readline_3.ps
    readline-6.2/doc/history_3.ps
    readline-6.2/doc/history.pdf
    readline-6.2/doc/readline.pdf
    readline-6.2/doc/rluserman.pdf
    readline-6.2/examples/
    readline-6.2/examples/autoconf/
    readline-6.2/examples/autoconf/BASH_CHECK_LIB_TERMCAP
    readline-6.2/examples/autoconf/RL_LIB_READLINE_VERSION
    readline-6.2/examples/autoconf/wi_LIB_READLINE
    readline-6.2/examples/rlfe/
    readline-6.2/examples/rlfe/ChangeLog
    readline-6.2/examples/rlfe/Makefile.in
    readline-6.2/examples/rlfe/README
    readline-6.2/examples/rlfe/config.h.in
    readline-6.2/examples/rlfe/configure
    readline-6.2/examples/rlfe/configure.in
    readline-6.2/examples/rlfe/extern.h
    readline-6.2/examples/rlfe/os.h
    readline-6.2/examples/rlfe/pty.c
    readline-6.2/examples/rlfe/rlfe.c
    readline-6.2/examples/rlfe/screen.h
    readline-6.2/examples/Makefile.in
    readline-6.2/examples/excallback.c
    readline-6.2/examples/fileman.c
[... 148 lines cut ...]
    creating build
    creating build/temp.linux-aarch64-3.7
    creating build/temp.linux-aarch64-3.7/Modules
    creating build/temp.linux-aarch64-3.7/Modules/3.x
    aarch64-openwrt-linux-gnu-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -O2 -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -O2 -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -DNDEBUG -fno-inline -O2 -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -DNDEBUG -fno-inline -I/media/ware/Entware.2019.10/staging_dir/target-aarch64_cortex-a53_glibc-2.27/opt/include -I/media/ware/Entware.2019.10/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.4.0_glibc-2.27/include -fPIC -DHAVE_RL_CALLBACK -DHAVE_RL_CATCH_SIGNAL -DHAVE_RL_COMPLETION_APPEND_CHARACTER -DHAVE_RL_COMPLETION_DISPLAY_MATCHES_HOOK -DHAVE_RL_COMPLETION_MATCHES -DHAVE_RL_COMPLETION_SUPPRESS_APPEND -DHAVE_RL_PRE_INPUT_HOOK -I. -I/opt/include/python3.7 -c Modules/3.x/readline.c -o build/temp.linux-aarch64-3.7/Modules/3.x/readline.o -Wno-strict-prototypes
    In file included from /opt/include/python3.7/Python.h:99:0,
                     from Modules/3.x/readline.c:8:
    Modules/3.x/readline.c: In function 'on_completion':
    /opt/include/python3.7/unicodeobject.h:1166:29: warning: initialization discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
     #define _PyUnicode_AsString PyUnicode_AsUTF8
                                 ^
    Modules/3.x/readline.c:842:23: note: in expansion of macro '_PyUnicode_AsString'
                 char *s = _PyUnicode_AsString(r);
                           ^~~~~~~~~~~~~~~~~~~
    Modules/3.x/readline.c: In function 'PyInit_readline':
    Modules/3.x/readline.c:1179:34: warning: assignment from incompatible pointer type [-Wincompatible-pointer-types]
         PyOS_ReadlineFunctionPointer = call_readline;
                                      ^
    creating build/lib.linux-aarch64-3.7
    aarch64-openwrt-linux-gnu-gcc -shared -L/media/ware/Entware.2019.10/staging_dir/target-aarch64_cortex-a53_glibc-2.27/opt/lib -Wl,-rpath,/opt/lib -Wl,-rpath-link=/media/ware/Entware.2019.10/staging_dir/target-aarch64_cortex-a53_glibc-2.27/opt/lib -Wl,--dynamic-linker=/opt/lib/ld-linux-aarch64.so.1 -L/media/w [... cut ...]
ERROR: Command errored out with exit status 1: /opt/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/opt/tmp/pip-install-iwwqg0oj/readline/setup.py'"'"'; __file__='"'"'/opt/tmp/pip-install-iwwqg0oj/readline/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /opt/tmp/pip-record-2ccdnu7n/install-record.txt --single-version-externally-managed --compile Check the logs for full command output.

listenfree, 2019-11-26:
> You can add it yourself - https://github.com/Entware/Entware/wiki/Self-installation-of-python-modules

would you please help me,thanks

zyxmon, 2019-11-26:
We do not suggest this sort of support.  Sorry.

Ask openwrt developers to add this package. We will add it in Entware after openwrt.

closed 2019-11-26 by zyxmon