Entware/Entware #372: locales install failed, download url incorrect

comicfans, 2019-11-27:
Hello, I'm running entware on R7000 with tomatousb

kernel 2.6.36.4brcmarm

while install opkg , locales package failed with error:

```
root@unknown:/opt/usr/lib/locale# opkg install locales --force-reinstall
Removing package locales from root...
Installing locales (2.23-8) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/locales_2.23-8_armv7-2.6.ipk
Configuring locales.
Entware uses separate locale-archive file independent from main system
Creating locale archive - /opt/usr/lib/locale/locale-archive
Adding en_EN.UTF-8
cannot map archive header: Invalid argument
Adding ru_RU.UTF-8
cannot map archive header: Invalid argument
locale-archive file was not created, not enough memory? Downloading...
checking endianess....
Byte Order: little endian.
Connecting to bin.entware.net (104.27.176.50:80)
wget: server returned error: HTTP/1.1 404 Not Found
You can downlo
[out.log](https://github.com/Entware/Entware/files/3895938/out.log)
ad locale sources from http://bin.entware.net/other/i18n_glib223.tar.gz
You can add new locales to Entware using /opt/bin/localedef.new

``` 

I've taken command from postinst file

```bash
/opt/bin/localedef.new -c -f UTF-8 -i en_US en_US.UTF-8
```

gives exactly same error. strace given following log:

[out.log](https://github.com/Entware/Entware/files/3895942/out.log)

and the download url is also incorrect, should be 227, not 223

zyxmon, 2019-11-27:
The download url should be 223 because yor feed uses ald kernel and glibc 2.22
It is fixed now - please try again and report.

Thanks for reporting this issue

comicfans, 2019-11-30:
confirm fixed

closed 2019-11-30 by comicfans