Entware/Entware #373: Support of busybox utilities in installation script

sat-neko, 2019-11-27:
Hello.

I have a E8372h-153 modem with custom firmware (see info attached [modem device info.txt](https://github.com/Entware/Entware/files/3897037/modem.device.info.txt)). I have tried to install entware on this modem and not succeed with execution of installation scripts. (I did it in telnet shell.)

After some investigation I found out that these scripts are incompatible with the modem environment. Incompatibilities I have discovered by my own are the following:
1) Environment contains wget 
only as a sub-command of busybox, there is no wget binary. It makes downloading of libraries and opkg (that happens after "Info: Opkg package manager deployment..." message ) unsuccessful. I was able to do it by modem's busybox, so generally it is achievable despite busybox wget syntax is different comparing with usual one.
2) It looks like downloaded version of opkg rely on tar and gzip binaries, that are again available only as busybox sub-commands and have another syntax.

Is there any way to make it work out of the box? Are there compilation options of opkg so it use busybox versions of applications?

After the investigation I have figured out that /opt catalogue within which I have operated is in tmpfs. It resulted in OOM. How do I verify if I can use modem's SD card block device as /opt storage?

My final aim is to run dnsmasq and siproxd. If I will be able to run strongswan with satisfactory performance, it will be perfect.

Any advice is appreciated.

zyxmon, 2019-11-27:
Just modify your PATH variable so that modem busybox is first in the PATH. Make a symlink or remount sdcard so /opt is pointing to sdcard.

sat-neko, 2019-11-27:
Let me clarify my issue a little bit.
1. I have a root shell on the modem (Putty telnet).
2. I have downloaded the standard installation script from [bin.entware.net](http://bin.entware.net/armv7sf-k3.2/installer/generic.sh) to my Windows PC
3. I copy-paste it to Putty line-by-line, and at line 27 I get the following
```
root@android:/ # wget $URL/opkg -O /opt/bin/opkg
/bin/sh: wget: not found
```

4. Despite that I can use busybox wget, like this:
`/system/bin/busybox wget -g -v -r "/armv7sf-k3.2/installer/opkg" bin.entware.net -A 104.27.176.50 -l /opt/bin/opkg`
It does not helps to complete the installation a lot because opkg binary executed at line 43 of generic.sh calls wget binary which is not contained in the environment.
`/opt/bin/opkg update`

**My first question** is how to make opkg binary to use busybox wget which has another command line syntax. The last fact means I can not just make a symlink like `ln -s /system/bin/busybox /bin/wget` to make it work.
```
root@android:/ # wget $URL/opkg -O /opt/bin/opkg
wget: invalid option -- 'O'
BusyBox v1.21.1 (2014-09-09 21:06:27 CST) multi-call binary.

Usage: wget [OPTION]... HOSTwget download and upload a file via HTTP

Options:
        -g, --download         Download
        -s, --upload           Upload
        -v, --verbose          Verbose
        -u, --username         Username to be used
        -p, --password         Password to be used
        -l, --local            Local file path
        -r, --remote           Remote file path
        -P, --port             Port to be used, optional
        -B, --bind             Bind local ip, optional
        -A, --addr             Remote resolved ip, optional
        -b, --begin            Transfer start position
        -e, --len              Transfer length
        -m, --max              Max transfer size
        -c, --compress         Compress downloaded file

243|root@android:/ # busybox which wget
/bin/wget
root@android:/ # echo $URL
http://bin.entware.net/armv7sf-k3.2/installer

```

I have just one busybox in the modem environment, and it is already in the PATH.
```
root@android:/ # find / -name busybox
/system/bin/busybox
root@android:/ # echo $PATH
/bin:/sbin:/vendor/bin:/system/sbin:/system/bin:/system/xbin:/app/bin:/bin:/system/bin/wifi_brcm/exe
root@android:/ # /system/bin/busybox which busybox
/system/bin/busybox
root@android:/ #
```

zyxmon, 2019-11-27:
Install this busybox - https://play.google.com/store/apps/details?id=ru.meefik.busybox&hl=en
It is known to be compatible with Entware installation.
Installing Entware on Android is always a little quiz. You can read some solutions in my blog (Russian) - http://www.zyxmon.org/

closed 2019-11-27 by zyxmon

sat-neko, 2019-12-03:
Thank you for the recommendation, this busybox binary works well.
JFTR, The next issue I hit is EXT2 filesystem is not supported by my device, but is it unrelated to Entware.
Thank you for your help, @zyxmon.