Entware/Entware #374: rclone Version

gelsas, 2019-11-29:
Hello, how can I update rclone manually when installed through entware ?

The latest rclone version is 1.50.2 while entware still has rclone 1.46.1.

zyxmon, 2019-11-29:
You can try to replace Entware binary with a new one from the rclone author. If it does not work - wait for the new version in Entware.

zyxmon, 2020-01-17:
rclone is updated - https://github.com/Entware/entware-go/blob/master/rclone/Makefile#L12
binaries will be available soon.

gelsas, 2020-01-27:
hello @zyxmon, just tried the following.

[~] # opkg update
Downloading http://bin.entware.net/x64-k3.2/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/entware
[~] # opkg upgrade
[~] # opkg install rclone
Package rclone (1.48.0-1) installed in root is up to date.
[~] #

Do i need to do something else to update rclone ?

regards

zyxmon, 2020-01-27:
>> Do i need to do something else to update rclone ?

Just wait - we will update binaries soon. 
You can also update x64 version manually - http://test.entware.net/x64-k3.2/rclone_1.50.2-1_x64-3.2.ipk

gelsas, 2020-09-09:
Hi,

rclone had its next big release: https://forum.rclone.org/t/rclone-1-53-release or https://rclone.org/downloads/

Could rclone get updated to the latest version in entware ?

zyxmon, 2020-09-09:
Already updated via https://github.com/Entware/entware-go/commit/1703221044e6c44bf99aefc18d96714e9e9ca46a
Binaries will arrive with Entware update.