Entware/Entware #375: Cannot find `util-linux`

acheronfail, 2019-12-03:
I can't seem to install/find the `util-linux` package, and according to what I can see online it seems that it _should_ be available:

* https://github.com/Entware/Entware/issues/51#issuecomment-447624363
* https://github.com/openwrt/openwrt/tree/master/package/utils/util-linux

(I might be completely wrong, though.)

However, this is what I'm getting when I try to install:

```bash
$ opkg update
Downloading http://bin.entware.net/x64-k3.2/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/entware
```

```bash
$ opkg install util-linux
Unknown package 'util-linux'.
Collected errors:
 * opkg_install_cmd: Cannot install package util-linux.
```

zyxmon, 2019-12-03:
List of Entware packages with search function - http://bin.entware.net/x64-k3.2/Packages.html
util-linux is split into a lot of small packages (openwrt style) - https://github.com/Entware/Entware/blob/master/package/utils/util-linux/Makefile#L809

closed 2019-12-03 by zyxmon

acheronfail, 2019-12-03:
Thanks for the links!

Ah, I was looking explicitly for `rev` but it seems it’s not included. 😔

igorpupkinable, 2023-02-21:
I am also looking for `rev`.
@acheronfail have you been able to find `rev` alternative or anything else?

acheronfail, 2023-02-22:
@igorpupkinable no, I ended up writing a python script to do the job if I remember correctly!

igorpupkinable, 2023-09-02:
`rev` utility is available as of util-linux version 2.28.1 release.