Entware/Entware #379: ghostscript package missing

redcitybass, 2019-12-13:
Hello! I'm trying to set up CUPS with entware on OpenWRT, but when i'm trying to install cups-pdf, I get an error:

`root@OpenWrt:/# epkg install cups-pdf
Installing cups-pdf (3.0.1-1) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/cups-pdf_3.0.1-1_armv7-3.2.ipk
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for cups-pdf:
 * 	ghostscript
 * opkg_install_cmd: Cannot install package cups-pdf.
root@OpenWrt:/# 
`

ryzhovau, 2019-12-13:
You can't use OpenWrt and Entware packages mutually.

closed 2019-12-13 by ryzhovau

zyxmon, 2019-12-13:
ghostscript package was corrupted, reuploaded it.
This should fix the problem.

redcitybass, 2019-12-13:
Now it can be installed, thanks!